mne-tools/mne-python #4893: Drop bad channels/ epochs. Gives RuntimeError: bad channel(s) ['Fp1'] marked do not exist in info

Stoffander, 2018-01-25:
Hey guys,

want to mark channel 'Fp1' as bad, which works perfectly, but then afterwards when I call the mne.pick_types function, I get the RuntimerError: bad channel(s) ['Fp1'] marked do not exist in info
```
# drop Fp1 channel
raw.info['bads'] = ['Fp1']

#work with eeg date, so eeg=True
picks_eeg = mne.pick_types(raw.info, meg=False, eeg=True, eog=False,
                           stim=False, exclude='bads')
```

Any idea what might the problem be? 

Thanks in advance!!

agramfort, 2018-01-25:
what does

print(raw.info)
print(raw.ch_names)

tell you?

Stoffander, 2018-01-25:
What I put in
```
#Input
# drop Fp1 channel
raw.info['bads'] = ['Fp1']

print(raw.info)
print(raw.ch_names)
```
What I got out
```
#Output:
<Info | 18 non-empty fields
    bads : list | Fp1
    buffer_size_sec : float | 1.0
    ch_names : list |  P7,  P4,  Cz,  Pz,  P3,  P8,  O1,  O2,  T8, ...
    chs : list | 21 items (EEG: 20, STIM: 1)
    comps : list | 0 items
    custom_ref_applied : bool | False
    dev_head_t : Transform | 3 items
    dig : list | 19 items
    events : list | 0 items
    highpass : float | 0.0 Hz
    hpi_meas : list | 0 items
    hpi_results : list | 0 items
    lowpass : float | 250.0 Hz
    meas_date : ndarray | 1970-01-01 00:00:00 GMT
    nchan : int | 21
    proc_history : list | 0 items
    projs : list | 0 items
    sfreq : float | 500.0 Hz
    acq_pars : NoneType
    acq_stim : NoneType
    ctf_head_t : NoneType
    description : NoneType
    dev_ctf_t : NoneType
    experimenter : NoneType
    file_id : NoneType
    hpi_subsystem : NoneType
    kit_system_id : NoneType
    line_freq : NoneType
    meas_id : NoneType
    proj_id : NoneType
    proj_name : NoneType
    subject_info : NoneType
    xplotter_layout : NoneType
>
[u' P7', u' P4', u' Cz', u' Pz', u' P3', u' P8', u' O1', u' O2', u' T8', u' Fpz', u' PO4', u' F4', u' Fp2', u' FC1', u' CP5', u' F3', u' Fp1', u' T7', u' FC2', u' PO3', 'STI 014']
```

larsoner, 2018-01-25:
The channel is named `' Fp1'` not `'Fp1'` (note the leading space)

Stoffander, 2018-01-25:
It worked. Thanks a lot!!! Seems as it was less a hard nut to crack then I thought haha

closed 2018-01-25 by Stoffander

mmagnuski, 2018-01-25:
@Stoffander You'd probably want to rename your channels to remove the spaces. You can use [`raw.rename_channels`](https://martinos.org/mne/stable/generated/mne.rename_channels.html):
```python
remove_blanks = lambda s: s.replace(' ', '')
raw.rename_channels(remove_blanks)
```