mne-tools/mne-python #4896: Mixed source space, pick_ori='vector', shape mismatch

SherazKhan, 2018-01-25:
I am trying to get vector time courses in mixed source space, got mismatch error. 
I think stc is initialized as scalar, causing shape mismatch.

```
import os.path as op
import numpy as np
import mne
import matplotlib.pyplot as plt

from mne.datasets import sample
from mne import setup_volume_source_space, setup_source_space
from mne import make_forward_solution
from mne.io import read_raw_fif
from mne.minimum_norm import make_inverse_operator, apply_inverse_epochs
from mne.connectivity import spectral_connectivity
from mne.viz import circular_layout, plot_connectivity_circle

# Set dir
data_path = sample.data_path()
subject = 'sample'
data_dir = op.join(data_path, 'MEG', subject)
subjects_dir = op.join(data_path, 'subjects')
bem_dir = op.join(subjects_dir, subject, 'bem')

# Set file names
fname_aseg = op.join(subjects_dir, subject, 'mri', 'aseg.mgz')

fname_model = op.join(bem_dir, '%s-5120-bem.fif' % subject)
fname_bem = op.join(bem_dir, '%s-5120-bem-sol.fif' % subject)

fname_raw = data_dir + '/sample_audvis_filt-0-40_raw.fif'
fname_trans = data_dir + '/sample_audvis_raw-trans.fif'
fname_cov = data_dir + '/ernoise-cov.fif'
fname_event = data_dir + '/sample_audvis_filt-0-40_raw-eve.fif'

# List of sub structures we are interested in. We select only the
# sub structures we want to include in the source space
labels_vol = ['Left-Amygdala',
              'Left-Thalamus-Proper',
              'Left-Cerebellum-Cortex',
              'Brain-Stem',
              'Right-Amygdala',
              'Right-Thalamus-Proper',
              'Right-Cerebellum-Cortex']

# Setup a surface-based source space
src = setup_source_space(subject, subjects_dir=subjects_dir,
                         spacing='oct6', add_dist=False)

# Setup a volume source space
# set pos=7.0 for speed issue
vol_src = setup_volume_source_space(subject, mri=fname_aseg,
                                    pos=7.0,
                                    bem=fname_model,
                                    volume_label=labels_vol,
                                    subjects_dir=subjects_dir)
# Generate the mixed source space
src += vol_src

# compute the fwd matrix
fwd = make_forward_solution(fname_raw, fname_trans, src, fname_bem,
                            mindist=5.0,  # ignore sources<=5mm from innerskull
                            meg=True, eeg=False,
                            n_jobs=1)

# Load data
raw = read_raw_fif(fname_raw, preload=True)
noise_cov = mne.read_cov(fname_cov)
events = mne.read_events(fname_event)

# Add a bad channel
raw.info['bads'] += ['MEG 2443']

# Pick MEG channels
picks = mne.pick_types(raw.info, meg=True, eeg=False, stim=False, eog=True,
                       exclude='bads')

# Define epochs for left-auditory condition
event_id, tmin, tmax = 1, -0.2, 0.5
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=(None, 0), reject=dict(mag=4e-12, grad=4000e-13,
                                                    eog=150e-6))

# Compute inverse solution and for each epoch
snr = 1.0           # use smaller SNR for raw data
inv_method = 'dSPM'  # sLORETA, MNE, dSPM
parc = 'aparc'      # the parcellation to use, e.g., 'aparc' 'aparc.a2009s'

lambda2 = 1.0 / snr ** 2

# Compute inverse operator
inverse_operator = make_inverse_operator(raw.info, fwd, noise_cov,
                                         depth=None, fixed=False)

stcs = apply_inverse_epochs(epochs, inverse_operator, lambda2, inv_method,
                            pick_ori='vector', return_generator=False)
```

**stack traceback**
```

Traceback (most recent call last):
  File "/autofs/space/megryan_001/users/sheraz/anaconda/lib/python2.7/site-packages/IPython/core/interactiveshell.py", line 2881, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-97-e7c6e47a4aa6>", line 93, in <module>
    pick_ori='vector', return_generator=False)
  File "<string>", line 2, in apply_inverse_epochs
  File "/autofs/space/megryan_001/users/sheraz/git/mne-python/mne/utils.py", line 728, in verbose
    return function(*args, **kwargs)
  File "/autofs/space/megryan_001/users/sheraz/git/mne-python/mne/minimum_norm/inverse.py", line 1131, in apply_inverse_epochs
    stcs = [stc for stc in stcs]
  File "/autofs/space/megryan_001/users/sheraz/git/mne-python/mne/minimum_norm/inverse.py", line 1067, in _apply_inverse_epochs_gen
    vector=(pick_ori == 'vector'), source_nn=source_nn)
  File "/autofs/space/megryan_001/users/sheraz/git/mne-python/mne/source_estimate.py", line 388, in _make_stc
    tstep=tstep, subject=subject)
  File "<string>", line 2, in __init__
  File "/autofs/space/megryan_001/users/sheraz/git/mne-python/mne/utils.py", line 728, in verbose
    return function(*args, **kwargs)
  File "/autofs/space/megryan_001/users/sheraz/git/mne-python/mne/source_estimate.py", line 2053, in __init__
    verbose=verbose)
  File "<string>", line 2, in __init__
  File "/autofs/space/megryan_001/users/sheraz/git/mne-python/mne/utils.py", line 728, in verbose
    return function(*args, **kwargs)
  File "/autofs/space/megryan_001/users/sheraz/git/mne-python/mne/source_estimate.py", line 481, in __init__
    'must match' % (n_src, data.shape[0]))
ValueError: Number of vertices (7847) and stc.shape[0] (23541) must match
```

agramfort, 2018-01-25:
thanks a lot for the bug report. can you have a lookt?

SherazKhan, 2018-01-25:
I will look, and update.

bloyl, 2018-01-27:
I was under the impression that `pick_ori='vector'` hadn't added support for volume or mixed source spaces. Has that changed? I'd love to be using it for discrete Volume Source spaces.

agramfort, 2018-01-27:
yes I think it still needs to be done ...

agramfort, 2018-08-04:
@SherazKhan is this still relevant? do you have time to look?

larsoner, 2021-01-07:
This should be fixed in `master`, we test mixed + vector in `test_source_estimate.py`

closed 2021-01-07 by larsoner