mne-tools/mne-python #4898: Verbosity control for _find_events

hekolk, 2018-01-26:
Hi,

it would be great if the ``@verbose`` decorator could be added (back) to ``mne.event._find_events`` (https://github.com/mne-tools/mne-python/blob/master/mne/event.py#L423 ):

The realtime epochs call this method during processing of each data chunk and using the default log level causes a lot of unnecessary console output for each stimulus. The relevant ``_process_raw_buffer`` method already uses the verbose parameter ( https://github.com/mne-tools/mne-python/blob/master/mne/realtime/epochs.py#L325 ), but the necessary decorator of ``_find_events`` got removed a while ago in commit c96eef4fef05b. Hence, it would be great if you could reinsert the decorator!

larsoner, 2018-01-26:
We should probably add the decorator, and not set `verbose='ERROR'` but rather let the user set it with the `find_events` argument to the constructor.

agramfort, 2018-01-26:
Fine with me

hekolk, 2018-01-27:
Thanks for the feedback - I created the mini-PR #4902, using verbose='ERROR' as default in the constructor argument.

hekolk, 2018-01-28:
thanks for merging!

closed 2018-01-28 by hekolk