mne-tools/mne-python #4899: Mixed source space

SherazKhan, 2018-01-26:
@agramfort @larsoner @dengemann Today I was talking with Matti, I am wondering how we normalized forward operator in case of mixed source space, say surface space with ico5 and 7mm for cerebellum volume source space. There should be a normalization factor ?

agramfort, 2018-01-27:
currently if you have a mixed source space "I think" you cannot use surface
ori and loose param. Everything should be free ori.

but **yet** this needs to be investigated and validated with simulations /
experiments.

SherazKhan, 2018-01-27:
I am currently working on mixed source space problem, I agrees, we need to verify/evaluate the bias between these two surfaces.

agramfort, 2018-01-27:
>
> I am currently working on mixed source space problem
>

great. Open a WIP PR asap so we can play with you.

larsoner, 2019-08-21:
Closing since we have other issues open having to do with mixed source space priors

closed 2019-08-21 by larsoner