mne-tools/mne-python #4901: Control the point size when plotting source spaces

bloyl, 2018-01-27:
I'd like to have control over the point size used to plot source spaces in plot_bem. Currently each source is plotted with a size of 1.

What is the preferred way to do this?

I have a patch that works but it changes the api of plot_bem
from 
```
def plot_bem(subject=None, subjects_dir=None, orientation='coronal',
             slices=None, brain_surfaces=None, src=None, show=True):
```
to
```
def plot_bem(subject=None, subjects_dir=None, orientation='coronal',
             slices=None, brain_surfaces=None, src=None, src_point_size=1,
             show=True)
```
I like keeping the `show` keyword last, but the api change could potentially break code that pass everything by position - `plot_bem('sample', subjects_dir, 'coronal', None, None, src, True)`.

Is there a reasonable alternative?

agramfort, 2018-01-27:
you need to update here the call to scatter:

https://github.com/mne-tools/mne-python/blob/master/mne/viz/misc.py#L330

can you show how you would like it to look? maybe we can change the size
globally to make it nicer
without an extra param?

bloyl, 2018-01-27:
I have a working patch, that changes the api. I can do a pull request if that would help.

I think that a global update would potentially be better provided its relatively easy to configure. Ideally it would be great to expose the point size, color, and alpha used in the scatter call.

If you give me an idea where I should configure these options, I can take a look at it.

agramfort, 2018-01-27:
can you show how it looks like now and how you would like it to look?

bloyl, 2018-01-28:
Here is the current code with a point size of 1. It is pretty hard to locate the src location. worse the point doesnt scale with zooming so that doesn't help.
[image]

At a minimum it would be nice to be able to control the size 
[image]

or even better have control over all the options to scattter, such as color, marker, alpha etc.
[image]

agramfort, 2018-01-28:
I am -1 adding so much parameters to the API.

See for example:

https://gist.github.com/agramfort/2b623b39d33f416cb1fae3c51f105a0e

it shows you how to edit a rendered figures taking the fig object returned
by the plotting function.

it should allow you to make any customization you need.

bloyl, 2018-01-31:
That works for me.

closed 2018-01-31 by bloyl