mne-tools/mne-python #4904: [RFC] crop behaviour

massich, 2018-01-29:
I was playing with crop and I found weird that based on this description the raw object gets modified:
```
Signature: reference_mne_data.crop(tmin=0.0, tmax=None)
Docstring:
Crop raw data file.

Limit the data from the raw file to go between specific times. Note
that the new tmin is assumed to be t=0 for all subsequently called
functions (e.g., time_as_index, or Epochs). New first_samp and
last_samp are set accordingly.

Parameters
----------
tmin : float
    New start time in seconds (must be >= 0).
tmax : float | None
    New end time in seconds of the data (cannot exceed data duration).

Returns
-------
raw : instance of Raw
    The cropped raw object.
```

Then going through the snipped provided in #4697 I realized that I might not be the only one having issues with it. Take a look at the following user code:

```
old_raw = raw.copy()
new_raw = raw.copy().crop(raw.times[times_start[0]],
                          raw.times[times_end[0]])

for idx in range(1, len(times_start)):
    cropped_raw = raw.copy().crop(raw.times[times_start[idx]],
                                  raw.times[times_end[idx]])
    mne.concatenate_raws([new_raw, cropped_raw])
```
either `old_raw` or `raw.copy()` should be unnecessary. 

So do you think we should change crop behavior or it's better to leave it as it is to keep backward compatibility?

dengemann, 2018-01-29:
We can improve the documentation and make explicit that it returns self, that is the modified raw object.

agramfort, 2018-01-29:
yes I think that this line:

old_raw = raw.copy()

is unnecessary. raw.crop indeed returns self.

larsoner, 2018-04-04:
Closing as addressed, feel free to reopen if I'm wrong @massich

closed 2018-04-04 by larsoner

agramfort, 2018-04-04:
I don't think it is fixed no?

larsoner, 2018-04-04:
Right, #5094 should do it

PR #5094 (2018-04-04): DOC: Docstring [ci skip]
Closes #4904.