mne-tools/mne-python #4914: plot_alignment fails

gvelchuru, 2018-02-01:
I have the following code:

`import mne
import numpy as np
from mayavi import mlab
from mne import Evoked, evoked
from mne.viz import plot_alignment, snapshot_brain_montage
import matplotlib.pyplot as plt
from scipy.io import loadmat

evoked_arr = evoked.read_evokeds('test-ave.fif')
subjects_dir = mne.datasets.sample.data_path() + '/subjects'

fig = plot_alignment(evoked_arr[0].info, subject='sample', subjects_dir=subjects_dir, surfaces=['pial'])
mlab.view(200, 70)
info = evoked_arr[0].info
xy, im = snapshot_brain_montage(fig, info)

xy_pts = np.vstack(xy[ch] for ch in info['ch_names'])

activity = np.linspace(100, 200, xy_pts.shape[0])

fig2, ax = plt.subplots(figsize=(10, 10))
ax.imshow(im)
ax.scatter(*xy_pts.T, c=activity, s=200, cmap='coolwarm')
ax.set_axis_off()
fig2.savefig('./brain.png', bbox_inches='tight')
`

plot_alignment creates a blank "Mayavi scene window", and if I keep continuing I get the following error:
X Error of failed request:  GLXBadCurrentWindow
  Major opcode of failed request:  154 (GLX)
  Minor opcode of failed request:  103 (X_GLXDeleteLists)
  Serial number of failed request:  467
  Current serial number in output stream:  470

My output ends up being
[image]
which has (as far as I can tell) the correct electrode positions but obviously no brain.

agramfort, 2018-02-01:
is it the code from example?

if so it's a mayavi install pb :(

gvelchuru, 2018-02-01:
It is the example code - any debugging things I can do or alternatives to mayavi?

agramfort, 2018-02-01:
how did you install mayavi?

can you run any mayavi example outside of mne?

gvelchuru, 2018-02-03:
I think this was an X forwarding issue as I was ssh-d into my computer, fixed by switching to my own computer

closed 2018-02-03 by gvelchuru