mne-tools/mne-python #4917: Marking ischemic damage on topoplots

suntracks, 2018-02-02:
Would like to know if there is a way to mark ischemic damage on topoplots.
It be great to visualise changes in evoked potentials around the damaged area of brain.

agramfort, 2018-02-03:
can you share an example figure to see what you have in mind?

suntracks, 2018-02-03:
Suppose we know that ischemic damage was in the region between P3 and P4, if  we could mark all the plots with this (maybe as a 'X' or 'O' or '.')  , so that we can view all the topo plots with the damaged region as a reference.

agramfort, 2018-02-03:
change channel names in data and plot channel locations with name in your
topo?

suntracks, 2018-02-03:
how can it be achieved in function for a series for plots starting t0 - t1

evoked.plot_topomap( ch_type='eeg',times=times)

agramfort, 2018-02-03:
use show_names and sensors options in plot_topomap

suntracks, 2018-02-08:
Thanks for the input, 
This really does not solve the purpose but a good work around.
is it possible to mark just one sensor position rather than all the channels from the list.

agramfort, 2018-02-08:
you mean a list with one element ? if so just put the channel name in []

suntracks, 2018-02-08:
This is showing up all the channel names specified in the channel name [].
plotting evoked data needs this info file marked correctly i suppose with all channel names & type.
Otherwise the montage locations are an issue. Correct me if am wrong here.

agramfort, 2018-02-09:
yes info needs to be complete to work.

we could allow show_names names to take a list of channels
to specify what channels to plot.

larsoner, 2018-02-09:
> we could allow show_names names to take a list of channels to specify what channels to plot.

I'd rather fix the dig sizing / position to properly assume head coords. Then this option is unnecessary as `pick_info` can be used.

For now a workaround would be to iterate over the text objects (using `matplotlib`) and change them based on their names (e.g., set some `visible=False` or change their colors).

jona-sassenhagen, 2018-02-26:
Could one also rename the channels in the evoked object used for plotting ..?

agramfort, 2018-08-04:
closing. @suntracks reopen if you think it's still relevant.

closed 2018-08-04 by agramfort