mne-tools/mne-python #4923: Importing Yokogawa/KIT data

makkostya, 2018-02-06:
I try to import data from Brainstorm tutorial: http://neuroimage.usc.edu/brainstorm/Tutorials/Yokogawa and get an error. However, I can import it in Brainstorm. My MNE version 0.15.2
I run this code:
```
import mne
mne.io.read_raw_kit('SEF_000-export.con')
```
output:
```
Extracting SQD Parameters from SEF_000-export.con...
Creating Raw.info structure...
```

error:
`IOError: Unknown FLL type: 300`

larsoner, 2018-02-06:
@christianbrodbeck any idea how we add support for this site?

christianbrodbeck, 2018-02-06:
Implementation of the SQD reader is dependent on information on the file format, which is unfortunately not complete. It looks like the relevant component of brainstorm is compiled without source code (https://github.com/brainstorm-tools/brainstorm3/tree/master/external/yokogawa).

PR #5077 (2018-04-03): MRG: Allow unknown KIT systems
Closes #4923.

Running this code on the Brainstorm KIT example data:
```
>>> import mne
>>> data_path = '/home/larsoner/mne_data/sample_yokogawa/data'
>>> raw = mne.io.read_raw_kit(data_path + '/SEF_000-export.con', preload=True)
RuntimeWarning: Unknown site filter settings (FLL) for system "Meg160/Analysis" model "PQA160C" (ID 1001), will assume FLL 300->200, check your data for correctness!
>>> raw.plot_psd()
RuntimeWarning: Channel locations not available. Disabling spatial colors.
```
Produces:
[image]

This might be good enough for now, until we find a bette [... cut]
closed 2018-04-03 by larsoner