mne-tools/mne-python #4926: Move test data out of MNE package

cbrnr, 2018-02-07:
I think it would be nice if we moved all test data out of `mne.io` and integrated them into `mne.datasets.testing` instead. Currently, the following subfolders of `mne/io` contain test files:

- brainvision/tests/data (1.1 MB)
- bti/tests/data (3.1 MB)
- edf/tests/data (3.1 MB)
- egi/tests/data (0.3 MB)
- kit/tests/data (5.1 MB)
- nicolet/tests/data (0.03 MB)
- tests/data (64.2 MB)

This would reduce the package size by 76.9 MB.

larsoner, 2018-02-07:
These do not get included in the release, see 5MB binary:

https://pypi.python.org/pypi/mne/0.15.2

For the repo to save the size we'd need to purge / rewrite history, probably not worth it.

larsoner, 2018-02-07:
(at least to save the size when doing `git clone`)

cbrnr, 2018-02-07:
OK that's fine then. It's still a bit inconsistent that some test files are in the package (e.g. EDF/BDF) whereas others (e.g. GDF) are in the testing datasets module.

cbrnr, 2018-02-07:
And in the repo, why do we need to rewrite history? Why not just `git rm`?

larsoner, 2018-02-07:
Agreed it's inconsistent it is historical cruft.

If you `git rm` the files "go away" but the (restore-able) record of them stays in the `git` history unless you purge (or do a sufficiently shallow clone) so there is no gain.

cbrnr, 2018-02-07:
> Agreed it's inconsistent it is historical cruft.

So OK to leave this open in case someone wants to work on it?

You're right, the size of the repo doesn't change, but this is not such a big issue. I think consistency is more important.

larsoner, 2018-02-07:
Since we seem to be stuck with the existing size in the repo, the current situation does have one advantage: even without access to the testing data (which is now getting a bit large) more core functionality can be tested / played with than if you remove those files.

agramfort, 2018-02-07:
we should be migrating to :

https://web.gin.g-node.org/NeuralEnsemble/ephy_testing_data

it's a git annex repo so we can selectively pull certiain from the this
potentially big repo

cbrnr, 2018-02-07:
Yeah, trading consistency for flexibility is not a good option. The testing data is not only getting a bit large (approaching 1GB I think), but the download is also extremely slow. I just aborted because I didn't want to wait for two hours. If this repo is a better option than what we have now we should look into it.

cbrnr, 2018-02-26:
@agramfort this repo sounds promising - anyone up to taking care of the migration? I guess it's too late to put it in a GSOC project...

agramfort, 2018-02-26:
that's not enough for a GSOC. It's a day of work. But it still needs a
volunteer...

cbrnr, 2018-05-22:
It seems like git-annex never gained enough momentum and was more or less replaced by Git LFS, which people seem to like better (https://stackoverflow.com/questions/39337586/how-do-git-lfs-and-git-annex-differ). We should discuss which of these two options we want to use.

Furthermore, once we have decided for one option, we need to find a place to host our testing data. When using Git LFS, we might be able to stay with GitHub. Maybe it is even possible to keep the separate mne-testing-data repo and just convert it to Git LFS. Then we'd need to update how we fetch the data (because we should be able to fetch individual files then). Or we create a new folder in the mne-python repo and let that folder be managed by Git LFS.

I have never used Git LFS, so I have no idea what works best. Any comments/ideas?

larsoner, 2018-05-22:
No idea, haven't used it either

agramfort, 2018-05-22:
we have started this initiative here:
https://web.gin.g-node.org/NeuralEnsemble/ephy_testing_data

but it's still WIP...

cbrnr, 2018-05-22:
And it's based on git-annex. Not sure if we should go for Git LFS instead (which means that the data could remain on GitHub, no additional hosting required).

agramfort, 2018-05-22:
the one that makes it work decides what we use :)

cbrnr, 2018-05-22:
This sounds almost like a threat :smile: - luckily it's not that urgent since I now know how to make it work if I only want to run a test with a single file.

larsoner, 2023-10-02:
Given the age of this issue (~5 years), the lack of other complaints or issues with our test data infrastructure in the meantime, and successful workflows/processes we have put in place and used with `mne-testing-data` over the same time span I'm inclined to close this issue

closed 2023-10-02 by drammock