mne-tools/mne-python #4928: Use pad_start and pad_stop in find_events

iraquitan, 2018-02-08:
As `find_events` function uses the `_find_stim_steps(data, first_samp, pad_stop=0, merge=merge)` under the hood, I believe it should be interesting to use these two arguments as well. I recently started using the [EEG Motor Movement/Imagery Dataset](https://physionet.org/pn4/eegmmidb/) and they also have events to rest state, so their events are as follows:
[image]

As you can see, the events already start at a non-zero number, because the event 1 is the rest state. With the current `find_events` function, this first event will be skipped as the current behavior is to only use `pad_stop`. I believe setting `pad_start=0` as default won't affect much, as in the vast majority of datasets, events start at 0.

My suggestions are:
* adding these two arguments (`pad_start` and `pad_stop`) to the `find_events` function
* setting `pad_start=0` as default behavior

If this new behavior is ok I can open a PR.

larsoner, 2018-02-08:
Can you see if #4756 (in latest `master`) fixes the first sample behavior?

iraquitan, 2018-02-08:
will check that

iraquitan, 2018-02-08:
```python
>>> raw = mne.io.read_raw_edf(fpath, preload=True)
Extracting edf Parameters from /../eegmmidb/S001/S001R06.edf...
EDF file detected
Setting channel info structure...
Created Raw.info structure...
Reading 0 ... 19999  =      0.000 ...   124.994 secs...
Ready.
>>> events = find_events(raw, initia_event=False)
Trigger channel has a non-zero initial value of 1 (consider using initial_event=True to detect this event)
Removing orphaned offset at the beginning of the file.
29 events found
Event IDs: [1 2 3]
>>> events = find_events(raw, initia_event=True)
30 events found
Event IDs: [1 2 3]
(30, 3)
```

Works as expected @larsoner . Thanks.

Is this supposed to work in 0.16?

larsoner, 2018-02-08:
> Is this supposed to work in 0.16?

Yes once it is released it should work in 0.16

closed 2018-02-08 by larsoner