mne-tools/mne-python #4930: MNE-Python failing test_io_set from mne.io.eeglab.tests.test_eeglab

ngayraud, 2018-02-08:
Hello, I am also getting an error in the same test as issue  #4855 (but a different error): 

`    
    ================ FAILURES  ================
    _______________    test_io_set   _______________

    args = (), kwargs = {}
    def skipper_func(*args, **kwargs):
        """Skipper for normal test functions."""
        if skip_val():
            raise SkipTest(get_msg(f, msg))
        else:
>           return f(*args, **kwargs)

/user/ngayraud/home/anaconda2/lib/python2.7/site-packages/numpy/testing/decorators.py:147: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
mne/io/eeglab/tests/test_eeglab.py:234: in test_io_set
    np.array([0., 0., 0.]))
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

x_id = array([ True,  True,  True], dtype=bool), y_id = array([False, False, False], dtype=bool), hasval = 'nan'

    def chk_same_position(x_id, y_id, hasval='nan'):
        """Handling nan/inf: check that x and y have the nan/inf at the same
            locations."""
        try:
            assert_array_equal(x_id, y_id)
        except AssertionError:
            msg = build_err_msg([x, y],
                                err_msg + '\nx and y %s location mismatch:'
                                % (hasval), verbose=verbose, header=header,
                                names=('x', 'y'), precision=precision)
>           raise AssertionError(msg)
E           AssertionError: 
E           Arrays are not equal
E           
E           x and y nan location mismatch:
E            x: array([ nan,  nan,  nan])
E            y: array([ 0.,  0.,  0.])

/user/ngayraud/home/anaconda2/lib/python2.7/site-packages/numpy/testing/utils.py:704: AssertionError`

I checked the solution proposed in MRG: Fix loc #4860, where all `[0., 0., 0.]` are replaced by `np.nan`.

It seems that the test passes when the line 231-232

`
assert_array_equal(raw.info['chs'][i]['loc'][:3],
                            np.array([0., 0., 0.]))
`

is replaced by

`assert_array_equal(raw.info['chs'][i]['loc'][:3], [np.nan] * 3)`

agramfort, 2018-02-08:
we recently changed empty EEG locations to NaN rather than 0.

what version are you testing?

ngayraud, 2018-02-08:
version 0.16

agramfort, 2018-02-08:
indeed. I see the same error locally.

@larsoner any idea what this is not captured by travis?

larsoner, 2018-02-08:
I see it locally, too. It could be because we have a SkipTest to work around a NumPy 1.14.0 (which Travis uses) bug. PR welcome to change the test.

PR #4932 (2018-02-10): [MRG] Fix eeglab test
closes #4930
closed 2018-02-10 by agramfort