mne-tools/mne-python #4931: source vertices get filtered out even with mindist=0. on sample

agramfort, 2018-02-08:
Run this script to replicate. You will see that even if you have mindist=0. you loose some vertices.  I am quite surprised about this. @larsoner any idea?

```
import os
import numpy as np
import mne
from mne.datasets import sample

data_path = sample.data_path()
# the raw file containing the channel location + types
raw_fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
# The paths to Freesurfer reconstructions
subjects_dir = data_path + '/subjects'
subject = 'sample'
trans = data_path + '/MEG/sample/sample_audvis_raw-trans.fif'

info = mne.io.read_info(raw_fname)

resolution = 3
expected_dip_per_hemi = {3: 642, 4: 2562, 5: 10242}

if 0:  # use morphed source space
    src_fs = mne.setup_source_space(subject='fsaverage',
                                    spacing='ico%d' % resolution,
                                    subjects_dir=subjects_dir, add_dist=False)

    src = mne.morph_source_spaces(src_fs, subject_to='sample',
                                  subjects_dir=subjects_dir)
else:
    src = mne.setup_source_space(subject='sample',
                                 spacing='ico%d' % resolution,
                                 subjects_dir=subjects_dir, add_dist=False)
print(src)

conductivity = (0.3,)  # for single layer
model = mne.make_bem_model(subject='sample', ico=4,
                           conductivity=conductivity,
                           subjects_dir=subjects_dir)
bem = mne.make_bem_solution(model)
fwd = mne.make_forward_solution(raw_fname, trans=trans, src=src, bem=bem,
                                meg=True, eeg=False, mindist=0., n_jobs=2)

assert src[0]['inuse'].sum() == src[0]['vertno'].size
assert src[1]['inuse'].sum() == src[1]['vertno'].size
assert src[0]['inuse'].sum() == expected_dip_per_hemi[resolution]

# BOOM !
assert fwd['sol']['data'].shape[1] == (expected_dip_per_hemi[resolution] * 6)

fwd_fixed = mne.convert_forward_solution(fwd, surf_ori=True, force_fixed=True,
                                         use_cps=True)
assert fwd_fixed['sol']['data'].shape[1] == (expected_dip_per_hemi[resolution] * 2)
```

larsoner, 2018-02-08:
Yes:
```
1 source space point omitted because it is outside the inner skull surface.
```
You can see it's probably one of these two in the posterior occipital lobe:
```
from mayavi import mlab
mne.viz.plot_alignment(info, trans, 'sample', subjects_dir, 'inner_skull', 'meg', meg=False, eeg=False, src=src)
mlab.view(-55, 22, roll=-43)
```
[image]

larsoner, 2018-02-08:
The inner skull seems to be a bit too tight:
```
mne.viz.plot_alignment(info, trans, 'sample', subjects_dir, ['inner_skull', 'pial'], 'meg', meg=False, eeg=False, src=src)
mlab.view(-50, 90, roll=-90)
```
[image]

agramfort, 2018-02-08:
hum ok. I am surprised it's even a pb flash images and using the white
surface though.

feel free to close.

closed 2018-02-08 by larsoner

jjnurminen, 2021-04-23:
Hi guys @larsoner @agramfort 

I'm encountering an issue where the BEM model becomes inaccurate for some occipital sources of the 'sample' subject. Excluding sources with `mindist` seems to help, but I have a feeling that the model is still inaccurate for part of the occipital region.

I'm not really a BEM/segmentation expert. Does the above mean that the 'sample' subject has geometry glitches that could affect BEM accuracy?

larsoner, 2021-04-23:
@jjnurminen see https://github.com/mne-tools/mne-python/pull/8840#issuecomment-774059118

The plan is to make some functions to quantify BEM quality using the rule of thumb mentioned there, maybe it would help with your use case?

jjnurminen, 2021-04-27:
@larsoner actually, after some struggle, I traced the problem down to the precomputed BEM that comes with the `sample` subject. For some reason, that model becomes very inaccurate for field points below the brain (relatively close to negative z-axis).  The problem goes away if I recompute the BEM using mne-python functions.

I couldn't reproduce using this another sample subject from the HF-SEF dataset.

jjnurminen, 2021-04-27:
@larsoner https://github.com/mne-tools/mne-python/issues/9354