mne-tools/mne-python #4941: Covariance plots - Colorbar missing

ngayraud, 2018-02-16:
In version 0.16, the colorbar setting (True or False) of the covariance plot (plot_cov function in file /viz/misc.py) does not affect the result. It is in fact not used in the code at all. Also,  when one sets show_svd=True the plots are not aligned since they are two different figures. 

I made a branch with some code to correct these small problems. The function returns a single figure containing either 2 of 4 subfigures. The code passes all tests.

Here are the figures, produced by running the following line of plot_simulate_evoked_data example:

> f1(,f2) = cov.plot(info,exclude = 'bads',colorbar = True,  show_svd= True)

original (in branch master):

[image]
[image]

aligned, with colorbar:
[image]

Let me know what you think

agramfort, 2018-02-16:
indeed... can you open a PR?
thanks a lot

jona-sassenhagen, 2018-03-31:
Closed by #4942 I assume.

closed 2018-03-31 by jona-sassenhagen