mne-tools/mne-python #4944: eeglab latency to event without sfreq

agramfort, 2018-02-17:
when working on annotations to events I realize that the eeglab reader
convert `eeg.latency` to sample index with `int(latency)`

see: https://github.com/mne-tools/mne-python/blob/master/mne/io/eeglab/eeglab.py#L697

reading https://sccn.ucsd.edu/wiki/Chapter_03:_Event_Processing it seems that latency is in seconds to converting to events would require `sfreq`

@jona-sassenhagen @smgutstein  @mmagnuski thoughts? is this a bug?

agramfort, 2018-02-17:
have a look at https://github.com/mne-tools/mne-python/pull/4936

mmagnuski, 2018-02-17:
@agramfort - latency is stored in samples in eeglab. But then different guis and functions use sometimes ms, sometimes s and sometimes samples.

mmagnuski, 2018-02-17:
The only issue might be matlab vs python indexing (if that is not dealt with when reading latencies).

mmagnuski, 2018-02-17:
And it seems that it is not taken care of in _read_annotations_eeglab:
`onset = [event.latency for event in eeg.event]`

agramfort, 2018-02-17:
event.latency has digits after the commas in our test file so it does not suggest it's samples but more seconds.

can you investigate with one of your files?

mmagnuski, 2018-02-17:
The file might have been downsampled - then events are then represented as happening between the samples in some cases. I can test that later today but I am almost 100% sure these are samples. But I'll check indexing offsets.

agramfort, 2018-02-17:
please have a look as there is something currently maybe broken.

mmagnuski, 2018-02-19:
I confirm that eeglab latency is in samples and that there is a one sample error caused by 0-based indexing in python and 1-based indexing in matlab. I can open a PR in the evening.

As I earlier mentioned - after resampling eeglab event latencies are often between the samples so `latency` becomes a float, not int. We should take care of this too - if currently we just cast these values to int, we can have other single sample errors. When `latency` is for example 2015.8 and it is cast to 2015 integer (or rather - 2014, given indexing differences between matlab and python :) ).

closed 2018-02-21 by larsoner

mmagnuski, 2018-02-21:
There are some points mentioned here that are left to adress - one of which is dealing with float latencies. Should I open a separate issue about that or should we leave this one open?

agramfort, 2018-02-21:
Open a new issue

PR #4951 (2018-02-21): [MRG+1]: fix event sample indices read from eeglab files
fix event latency from matlab 1-based to python 0-based.
Fixes #4944.