mne-tools/mne-python #4947: source leakage correction

dokato, 2018-02-18:
I don't think that MNE currently supports source leakage correction. According to the work of people from OHBA: https://www.ncbi.nlm.nih.gov/pubmed/27262239, correlation of envelopes after correction is one of the most reliable connectivity estimators.
For the purpose of my study I implemented symmetric orthogonalization from this paper: https://www.ncbi.nlm.nih.gov/pubmed/25862259
It's here: https://gist.github.com/dokato/7a997b2a94a0ec6384a5fd0e91e45f8b .
I'm happy to contribute it to mne if you want (but need some advice e.g. on where to place it).

dengemann, 2018-02-18:
Hi,

That sounds like a good idea. We plan adding power envelope correlations
soon. It’s essentially written but I need time to add it to MNE. Source
leakage correction should come with it but I haven’t looked into it yet.
Perhaps we can team up :)

Denis

dokato, 2018-02-18:
@dengemann Sure, sounds like a great idea! My implementation requires some amendments obviously, but in my tests worked fine. Let me know if you have any guidelines how to insert it to mne. My guess is that it should be a part of connectivity subpackage, shouldn't it? And then utils, or rather separate module (e.g. `leakage.py`)?

dengemann, 2018-02-18:
One reason we still don't have power envelope correlations in MNE is that I
would like to introduce a new API for this that is independent from the
spectral connectivity one. This always takes discussions, hence, time. I
think we should do a submodule for envelope correlations and just add a
leakage module there. Maybe I should start a PR next week to propose the
API I have in mind. Then we can take it from there.

dengemann, 2018-02-18:
Maybe we could hang out next week to exchange / synchronize ideas? It is
not unlikely that we have some parallel work going :)

dokato, 2018-02-18:
Yeah. Please, keep me in the loop about envelopes and I'm happy to hang out next week as well!

dengemann, 2018-02-18:
Cool. My email is denis.engemann@gmail.com, let's find a date for next week.

Denis

agramfort, 2018-02-20:
that should land in the connectivity module. It would be great if this code
enters
master with a tutorial on rest data eg using the brainstorm data we already
ship
or any other data source.

cc @SherazKhan

dengemann, 2018-02-20:
@agramfort I plan to work on that, finishing what we started in Boston last
year with Sheraz.

agramfort, 2018-08-04:
small reminder @dengemann :)

dengemann, 2018-08-06:
Depends on #5389.

larsoner, 2019-08-21:
We have envcorr now

closed 2019-08-21 by larsoner