mne-tools/mne-python #4948: Rounding bug in axis labels? Perhaps MPL?

jona-sassenhagen, 2018-02-18:
As of recently, something is weirding up some of the axis labels - rounding messes with the formatting. It's not an MNE issue per se- maybe a matplotlib update? Anyone got an idea?

Before:
[image]
Now:
[image]

mmagnuski, 2018-02-19:
Oh, I got that too some weeks ago. I didn't dig to find what causes these labels go weird.

agramfort, 2018-02-20:
please share a gist to replicate

jona-sassenhagen, 2018-02-20:
You can just do

```
import mne
from mne.datasets import testing

data_path = testing.data_path()
fname = data_path + "/EEGLAB/test_raw.set"
montage = data_path + "/EEGLAB/test_chans.locs"

event_id = {"rt":1, "square":2}
raw = mne.io.eeglab.read_raw_eeglab(fname, preload=True, eog=eog,
                                    montage=montage, event_id=event_id
                                   )
events = mne.find_events(raw)
epochs = mne.Epochs(raw, events, {"square":2})
evoked = epochs["square"].average()
mne.viz.plot_compare_evokeds(evoked);
```

And you get

[image]

mmagnuski, 2018-02-20:
But I guess it doesn't happen if you take that data and plot it by hand? Any chance this is due to how `plot_comapre_evokeds` passes ticks or ticklabels to plt.plot?

agramfort, 2018-02-23:
I don't have a pb with matplotlib 2.1.0

larsoner, 2018-02-23:
@jona-sassenhagen what is `mne.sys_info()`? It could be due to some NumPy update, too.

jona-sassenhagen, 2018-02-23:
```
mne.sys_info()
Platform:      Darwin-17.4.0-x86_64-i386-64bit
Python:        3.6.4 |Anaconda, Inc.| (default, Jan 16 2018, 12:04:33)  [GCC 4.2.1 Compatible Clang 4.0.1 (tags/RELEASE_401/final)]
Executable:    /Users/jona/miniconda3/bin/python
CPU:           i386: 4 cores
Memory:        16.0 GB

mne:           0.16.dev0
numpy:         1.14.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.0.0
matplotlib:    2.1.2

sklearn:       0.19.1
nibabel:       2.1.0
mayavi:        Not found
pycuda:        Not found
skcuda:        Not found
pandas:        0.22.0
```

jona-sassenhagen, 2018-02-23:
I'm gonna take a real look at what's happening now ... Just wanted to ask if anyone had any immediate ideas.

mmagnuski, 2018-02-23:
I recently had that problem with pure matplotlib (+seaborn) and it seemed to be due to NumPy as Eric suggests. I didn't explore further, however.

jona-sassenhagen, 2018-02-23:
If you check the PR, here, the problem is essentially doing `ax.set_xticklabels(ax.get_xticks())`.

PR #4965 (2018-02-24): MRG+1, MINOR fix float xtick labels in plot_compare_evoked
I don't know why I originally put that line there?

Can we test what it outputs to the screen? I can't think of a test.

closes #4948
closed 2018-02-24 by larsoner