mne-tools/mne-python #4949: undead vampire meas date

jasmainak, 2018-02-20:
I'm just borrowing terminology from chat with @dengemann for title :-)

I seem to have hit a bug when writing tests for `mne-bids`. Essentially, if you set `meas_date` to `None`, when writing the file, MNE is using the current date/time to set it. Therefore, when you read it back, it's not `None` anymore. I would consider it a bug but feel free to disagree :) To reproduce, run this:

```py
import os.path as op

import mne
from mne.datasets import sample
from mne.utils import _TempDir

data_path = sample.data_path()

raw_fname = op.join(data_path, 'MEG', 'sample', 'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(raw_fname)

# Let's make meas_date None
raw.info['meas_date'] = None
# raw.info['meas_id'] = None  # even uncommenting this does not change anything

# write it to a temp file
data_path2 = _TempDir()
raw_fname2 = op.join(data_path2, 'sample_audvis_raw.fif')
raw.save(raw_fname2, overwrite=True)

# and read it back
raw2 = mne.io.read_raw_fif(raw_fname2)
assert(raw.info['meas_date'] == raw2.info['meas_date'])
```

christianbrodbeck, 2018-02-20:
Sounds like that might be what prevented #2564 from working

jasmainak, 2018-02-20:
oh boy ...

agramfort, 2018-02-20:
yes it's a bug. We should not "guess" a meas_date.

can you send a PR?

closed 2018-04-13 by agramfort