mne-tools/mne-python #4950: BIDS compatibility

jasmainak, 2018-02-20:
Right now, there are a couple of compatibility issues in MNE with MEG-BIDS (draft). 

- The warning for `_raw.fif`should be removed and replaced by `_meg.fif`
- The anonymization should make the date before 1900, not 1970 (see main draft).
- Files greater than 2 GB should be written as `part-01_meg.fif` and `part-02_meg.fif`

Perhaps first one is easy issue for new contributor. Feel free to add/modify this list.

cc @teonbrooks @choldgraf @agramfort

agramfort, 2018-02-20:
1/ you mean we should not warn when file ends with _meg.fif ?

2/ regarding dates anonymization should make the date empty no? let's not
guess a date...

3/ regarding name splits be careful not to rename the files as the names
are in the fif files.

the line that should be changed is :
https://github.com/mne-tools/mne-python/blob/master/mne/io/base.py#L2166

I would like to avoid breaking API brutally...

teonbrooks, 2018-02-20:
> 1/ you mean we should not warn when file ends with _meg.fif ?

yeah, I think that would be a better solution: handle mne conventions like it does and allow for the bids naming conventions

> 2/ regarding dates anonymization should make the date empty no? let's not guess a date...

i think this will be fixed once we change the default reading of fif to keep `meas_id['secs']` and `meas_id['usecs']` values albeit zero (the no guessing implementation).

> 3/ regarding name splits be careful not to rename the files as the names are in the fif files.

I think the solution we came up with is that we would need to make a copy of fif files to be compatible for splitting. the tough consideration would be, how to change the split file pattern without having to write a new save function or exposing public keyword argument...

teonbrooks, 2018-10-17:
moving discussion from https://github.com/mne-tools/mne-bids/issues/12 here.

I updated @jasmainak's checklist above. did we resolve the anonymization step?

sappelhoff, 2019-04-24:
> I updated @jasmainak's checklist above. did we resolve the anonymization step?

No, I don't think so. At [anonymization](https://github.com/mne-tools/mne-python/blob/92f61faef72a335d6de63e816bd5bcfeaceffa09/mne/io/meas_info.py#L2008-L2038), `meas_date` is assigned `None`, and because we cannot have `None` as a date (as @larsoner [commented](https://github.com/mne-tools/mne-python/issues/4950#issuecomment-367095176)), this is actually a "magic date", see:

https://github.com/mne-tools/mne-python/blob/92f61faef72a335d6de63e816bd5bcfeaceffa09/mne/io/write.py#L20-L24

BTW: This is the part in the BIDS specification that says anonymized date should have a year <= 1900 ([link](https://bids-specification.readthedocs.io/en/stable/03-modality-agnostic-files.html#scans-file)). I quote:

> For anonymization purposes all dates within one subject should be shifted by a randomly chosen (but common across all runs etc.) number of days. This way relative timing would be preserved, but chances of identifying a person based on the date and time of their scan would be decreased. Dates that are shifted for anonymization purposes should be set to a year 1900 or earlier to clearly distinguish them from unmodified data.

sappelhoff, 2019-04-24:
Note: `DATE_NONE = (0, 2 ** 31 - 1) ` means `2147483647` microseconds, which are about 35 minutes.

So that refers to the 35th minute in year 0 (Jesus' death?) ... strictly speaking this fulfills the requirement:

> should be set to a year 1900 or earlier

So I guess we only remain with one bullet point from the original issue:

> The warning for _raw.fifshould be removed and replaced by _meg.fif

jasmainak, 2019-04-24:
haha, no I think `None` always points to 1st January 1970. Probably the mean of Alex's and Matti's bdays

sappelhoff, 2019-04-24:
Could you point to the code that specifies this? I only found the above, Jesus' death day

jasmainak, 2019-04-24:
well I can point to wikipedia: https://en.wikipedia.org/wiki/Unix_time :-)

sappelhoff, 2019-04-24:
oh, okay - thanks. I forgot what unix time is. So we could set:

`DATE_NONE = (0, 2 ** 31 - 1) `

to

`DATE_NONE = (-2208988800, 2 ** 31 - 1) `

Then it'd be 1st of January 1900, 35 minutes past midnight

sappelhoff, 2019-04-24:
After some discussion with @larsoner we resolved that the main work for anonymization of dates should be done in https://github.com/mne-tools/mne-bids --> namely the whole dataset with all `scans.tsv` files and raw data files that contain recording dates must be passed and a **consistent** constant must be deducted from the recording dates. (i.e., the same for all participants, tasks, sessions, runs, ...)

In MNE-Python we could add an argument `deduct_constant` to the anonymization code, which defaults to `0`, but can be set to some arbitrary integer to shift the date accordingly.

sappelhoff, 2019-06-05:
for reference, the file split naming according to BIDS (point 3 in the [OP](https://github.com/mne-tools/mne-python/issues/4950#issue-298439833) above)

was fixed:
- here: https://github.com/mne-tools/mne-python/pull/5470
- and here: https://github.com/mne-tools/mne-python/pull/5570

sappelhoff, 2022-06-05:
closing, we got this in mne-bids.

closed 2022-06-05 by sappelhoff