mne-tools/mne-python #4952: trial-dependent baseline

cristidonos, 2018-02-20:
Hi,
Current implementation of epochs.apply_baseline(..) is useful for data analysis in respect to stimulus onset. For analyzing data in respect to reaction times it would be useful to define an individual baseline for each trial/epoch. 
For example, i have a reading task where i use (-500, 0) ms pre-stimulus as baseline and (-500, 1000) centered on the reaction time (in my case speech articulation) as the region of interest. 
Is this possible with the current version of MNE? If not, do you consider adding this functionality?
Thanks,
Cristi

mmagnuski, 2018-02-21:
Currently you would not be able to do this even using `mne.baseline.rescale` but that's a reasonable feature request, we should think about adding support for operations like this to `apply_baseline`.

In the meantime, you still can get your hands dirty and do something like this:
```python
zero_idx = np.abs(epochs_stim.times - 0.).argmin()
baseline = epochs_stim._data[..., :zero_idx + 1].mean(axis=(0, -1), keepdims=True)
epochs_resp._data /= baseline
```

agramfort, 2018-02-21:
I think this has be mentioned by @christianbrodbeck or @teonbrooks in the past.

currently the only option is what @mmagnuski suggests.

jona-sassenhagen, 2018-02-21:
IIRC there's a paper somewhere by I think A Delorme where he shows single-trial baselining of time-frequency data is better.

cristidonos, 2018-02-21:
Thanks for the replies, i'll try the workaround @mmagnuski  suggested, with the caveat that trials will still need to be cropped based on reaction times after baseline correction. I'm still thinking what is the easiest way to do it, probably create 1 set of epochs containing only the baselines an another set of epochs centered on reaction times.
@jona-sassenhagen  : here's the paper https://www.ncbi.nlm.nih.gov/pubmed/21994498

mmagnuski, 2018-02-21:
> I'm still thinking what is the easiest way to do it, probably create 1 set of epochs containing only the baselines an another set of epochs centered on reaction times.

That's what my `epochs_stim` (epochs with respect to stimulus) and `epochs_resp` (epochs with respect to the response) stand for. :)

> here's the paper https://www.ncbi.nlm.nih.gov/pubmed/21994498

There is also another paper on that: [link](https://www.sciencedirect.com/science/article/pii/S1053811913009919). IIRC it disagrees with Grandchamp, Delorme on several points (for example the authors claim that subtraction baseline is better from percentage baseline).

mmagnuski, 2018-02-21:
@cristidonos BTW, in that piece of code I provided baseline from individual trials is averaged, you may want to use `.mean(axis=-1, keepdims=True)` if you don't want that.

jona-sassenhagen, 2018-03-03:
I'd like to tackle this.

- add a kwarg to `rescale` causing it to loop over trials when doing the rescaling
- add a kwarg to the `apply_baseline` methods to that regard

agramfort, 2018-03-04:
how would the code look like?

in a fif file the baseilne is bmin bmax so we won't be able to store the
baselines in a fif file without loosiing the info of how baseline was
computed

jona-sassenhagen, 2018-03-04:
Could store baseline in metadata ..?

agramfort, 2018-03-04:
yes that would work