mne-tools/mne-python #4953: BUG/ENH? no check for unique permutations in 2-samp clustering tests

dengemann, 2018-02-21:
Looking at the permutation mechanics in cluster_level.py it is striking that for 1samp tests we take care of exact permutations and avoid duplicates via ```_get_1samp_orders``` whereas for 2samp tests we simply rely on ```rng.permutations``` which does btw. not cycle through all permutations before repeating but just redraws permutations randomly. This can in principle lead to an inflated H0 when stats are repeated. Anyways I do not see any reasons for this forking of logic.

Any thoughts @larsoner ?

larsoner, 2018-02-21:
> Anyways I do not see any reasons for this forking of logic.

Deciding sign flips (one sample) goes as ~`2 ** n`, choosing permutations goes as ~`n!` so they will need to be different. But yes we shouldn't repeat if possible. PR welcome

dengemann, 2018-02-21:
@larsoner could we do something like this to precompute permutation indices for multiple samples, inspired by ```_get_1samp_orders```:

```Python

def _get_perms(n_permutations, sample_range, rng):

    permut_dict = {}
    while len(permut_dict) < n_permutations:
        this_permut = rng.choice(
            sample_range,
            len(sample_range),
            replace=False)
        if tuple(this_permut) not in permut_dict:
            permut_dict[tuple(this_permut)] = None
    permutations = np.array(list(permut_dict.keys()))
    return permutations

```

This seems to work reasonably fast and looks correct. It should be easy to insert this as a patch here:

https://github.com/mne-tools/mne-python/blob/master/mne/stats/cluster_level.py#L847

replacing the ```rng.permutations``` call, and iterating over the permutation indices instead.

Let me know what you think.

dengemann, 2018-02-28:
@larsoner you had something like this in mind?

```Python
def _get_perms_unique(n_permutations, master_index, rng, group_slices):
    permut_dict = {}
    while len(permut_dict) < n_permutations:
        this_permut = rng.choice(
            master_index,
            len(master_index),
            replace=False)
        test_sorted = [
            tuple(np.sort(master_index[this_group])) for this_group in
            group_slices]
        if tuple(test_sorted) not in permut_dict:
            permut_dict[tuple(this_permut)] = None
    permutations = np.array(list(permut_dict.keys()))
    return permutations
```
where grou_slices are a list of indexers to get the sub groups form the ```master_index```.

larsoner, 2018-03-01:
Yeah something like that. A simple test would be to make the lists small (two groups of two or three?) and ensure that the correct (hard-coded) groupings are generated when you ask for the full set of permutations.

Also there needs to be some protection against trying to generate more than are possible, as the current code would loop infinitely :)

kingjr, 2018-03-01:
> This can in principle lead to an inflated H0 when stats are repeated.

can you elaborate?

dengemann, 2018-03-01:
> can you elaborate?

@kingjr in principle it can go in any direction. But it could be that by chance you have n times an identical permutation that shows a high test statistic. In that case you would over estimate the probability of making your observing under H0. In the best case these things cancel out. Yet you would get an impoverished estimate as the true number of permutations would be the number of statistically unique permutations. So this is on the bug spectrum to me. Is it a bit clearer?

kingjr, 2018-03-01:
Mmmmh, I get the feeling, but I'm not sure. If the permutations are random, it shouldn't introduce a bias. But perhaps this could slightly mess up the precision of the p value estimate. If you have a clearer idea, could you make a quick simulation to show that when the two strategies differ (partition perms vs shuffle & redraw), they end with distinct p vals?

dengemann, 2018-03-01:
>  But perhaps this could slightly mess up the precision of the p value estimate.

I think this is the biggest risk. If you don't hit enough permutations, you may depend more on random seeds. You probably get unstable p-values.

dengemann, 2018-03-01:
> If you have a clearer idea, could you make a quick simulation to show that when the two strategies differ (partition perms vs shuffle & redraw), they end with distinct p vals?

Yep. A good call. I would expect the current code and the corrected code to meet once all possible permutations are exploited. The current code should stabilize later. I can't promise to find time soon but I'll look into it. These things clearly matter for #4859

kingjr, 2018-03-01:
>>    But perhaps this could slightly mess up the precision of the p value estimate.

> I think this is the biggest risk. If you don't hit enough permutations, you may depend more on random seeds. You probably get unstable p-values.

Ok, that's my feeling too. which means that it's not really an inflation of H0, but an imprecision; it's then publication bias that generate the inflation.

dengemann, 2018-03-01:
> Ok, that's my feeling too. which means that it's not really an inflation of H0, but an imprecision; it's then publication bias that generate the inflation.

Yes. You realize that with little permutations the seed matters more. More permutations, more precision. I think as providers of stats tools we want to be sure that people can control the effective number of permutations.

dengemann, 2018-03-01:
@kingjr Maybe we should make such simulations an example here. 1) Plot how p-value fluctuates and eventually stabilizes as you add permutations. 2) Show paths for different seeds and admire variance between them at lower number of permutations.

larsoner, 2018-04-02:
Pushing milestone to 0.17 but if you have time @dengemann feel free to tackle it now

dengemann, 2018-04-02:
Push to 0.17 ok

larsoner, 2018-10-04:
@dengemann is this bug still there? Can you fix it in the next two weeks? Otherwise we should move it to 0.18

larsoner, 2018-11-05:
Moving to 0.18