mne-tools/mne-python #4957: Q: how are we supposed to save beamformer filter

kingjr, 2018-02-21:
Once we have done 
```python
filters = mne.beamformer.make_lcmv()
```
h5io isnt applicable because of the EmpiricalCovariance object

larsoner, 2018-02-21:
Hmm, we shouldn't have an `EmpiricalCovariance` in there, just `Covariance` (which is internally a `dict` and thus h5io-compatible)

britta-wstnr, 2018-03-12:
I can put that on my agenda for when I take care of the deprecation of ``lcmv()`` (to make it comparable to the DICS API). The question is whether we want to wait for the big DICS PR to be merged first?

larsoner, 2018-03-12:
> The question is whether we want to wait for the big DICS PR to be merged first?

Yes to avoid having to rebase that one (it's big).

britta-wstnr, 2018-03-12:
@larsoner that's what I figured. I will take care of this as soon as the other one is merged then!

larsoner, 2018-04-02:
@britta-wstnr it looks like the changes to `lcmv.py` are pretty small in #5066. Do you have time to tackle saving/loading now? If so we can get probably get it into 0.16, because it should be quite simple with `h5io`. You can draft a PR for LCMV, we can get that up to snuff while #5066 is merged, then the addition of DICS should be just a dozen or so lines of code hopefully!

britta-wstnr, 2018-04-03:
I am confident that I can have a look before next week - also on the list is the deprecation of ``lcmv()`` to mimic the new DICS code, I can tackle that in the same go, then.

britta-wstnr, 2018-04-04:
So the filter contains the same covariance instance that is fed to ``make_lcmv`` (some changes might be done by ``pick_channels_cov``). In my case, this is always an ``Covariance`` object, and then everything saves fine. 
To be able to investigate this: how is the ``EmpiricalCovariance`` object obtained?

larsoner, 2018-04-04:
> To be able to investigate this: how is the EmpiricalCovariance object obtained?

Looking at the code I don't see why it would happen. @kingjr can you provide a snippet to replicate?

In the meantime @britta-wstnr you can proceed as if this is not an issue, assuming we will fix whatever is causing @kingjr to have the an EmpiricalCovariance rather than Covariance.

agramfort, 2018-08-04:
@kingjr I cannot replicate. 

Putting some of these lines in our tests do not crash:

```
        # Trying saving to disk
        mne.externals.h5io.write_hdf5('/tmp/tmp.h5', filters, overwrite=True)
```

larsoner, 2018-10-04:
I hit this bug at some point, basically if you compute the `cov` on the fly it can have `cov['estimator']` which is a `sklearn` object so you need to remove it before saving. I can tackle this I/O for 0.17

PR #5589 (2018-10-13): MRG: Add I/O and repr to beamformers
Closes #4957.

cc @wmvanvliet @britta-wstnr (and @kingjr since you opened the original issue)
closed 2018-10-13 by agramfort