mne-tools/mne-python #4959: API/ENH: allow to do pure maximum statistic permutation tests without clustering

dengemann, 2018-02-21:
Maximum statistic tests, like implemented in ```mne.stats.permutation_t_test``` are popular choices in M/EEG http://onlinelibrary.wiley.com/doi/10.1111/j.1469-8986.2011.01273.x/full
I feel we could easily make our clustering API perform such tests by skipping the clustering if ```connectivity=False```. This would allow us to use the same API to perform to perform general max stat tests for 1 and multiple samples with arbitrary stat functions, beyond the 1 sample t-test in ```mne.stats.permutation_t_test```. 

cc @agramfort @larsoner @SherazKhan etc.

agramfort, 2018-02-22:
permutation_t_test is here because it uses a numerical trick to make it
faster.
But I agree we could hide this from high level API.

allowing connectivtiy=False in :

permutation_cluster_test(X[, threshold, …])

permutation_cluster_1samp_test(X[, …])

is weird as you have cluster in the function names.

If we do this we should rename to:

permutation_test(X[, threshold, …])

permutation_1samp_test(X[, …])

or just add these functions and internally use the same private function.

larsoner, 2018-02-22:
> or just add these functions and internally use the same private function.

I am -1 on adding a new function that is identical other than having a `connectivity` argument. This proliferation of names would be counterproductive.

> permutation_t_test is here because it uses a numerical trick to make it faster.

Maybe we shouldn't make this function public, then. I have to imagine 1) it's not used that much, and 2) if it is, speed is *very* rarely a limiting factor. In the meantime every function we have in the API page that isn't in the 99% use case for stats for our users reduces API doc usability.

> If we do this we should rename to:

If we do rename, we should wait do it at the same time as #4859 so we can properly deprecate. In the meantime a stop-gap solution is to allow `connectivity=False` despite `cluster` being in the name. It would work as a sort of advanced use case without adding more API cruft.

jona-sassenhagen, 2018-02-23:
Erics suggestions make sense to me.

agramfort, 2018-02-23:
ok fine with me if it's temporary provided #4859
<https://github.com/mne-tools/mne-python/issues/4859> gets completed.

larsoner, 2018-02-23:
@mmagnuski would you have time to work on #4859 if we actually settle on some API?

mmagnuski, 2018-02-23:
I also agree with Eric's suggestions. In the long run it would be nice to just have `permutation_test`, unless it would require an internal mess or strange API. :)
I would be happy to work on #4859, but I am sure I won't have time for larger PRs in the next two months (I still have plans for multiple smaller PRs :) ). But I'll participate in API-discussions for sure!

dengemann, 2018-02-26:
> if it is, speed is very rarely a limiting factor.

For benchmarking Implemented a generalized version that takes any stat_fun. When running it with ```ttest_1samp_no_p``` the loss in performance is not game changing. Another issue that this implementation has is memory as all permutations are precomputed, which on larger data consumes most of the time. It can also be done with a generator. But I agree that we should recycle ```permutation_t_test```.

I agree with the common elements in Eric's and Alex' suggestions. Some temporary compromise would be ok. The next stable API should emerge with this GSOC.

PR #5096 (2018-04-19): MRG: Allow connectivity=False
Closes #4959.

@dengemann can you see if this works for your purposes?
closed 2018-04-19 by agramfort