mne-tools/mne-python #4961: EEG channels not found when creating EOG epochs

aitorme, 2018-02-23:
I'm trying to create an ICA template and after fitting the ICA I'm trying to get the eog epochs from `preprocessing.create_eog_epochs`, I get the following runtime error `RuntimeError: EEG 61 or EEG 62 channel not found !!`, however, my setup only has 32, and I use a "standard10_20" montage in order for all my channels to be matched. How can I change the amount of channels when running epoching functions?

agramfort, 2018-02-23:
do you have an EOG channel? What does

print(raw.info)

tell you?

aitorme, 2018-02-23:
I have 2 EOG channels I label myself, 
`bads : list |  EOG2,  EOG1
    buffer_size_sec : float64 | 1.0
    ch_names : list | Ch1, Ch2, Ch3, Ch4, Ch5, Ch6, Ch7, Ch8, Ch9, ...
    chs : list | 33 items (EEG: 30, STIM: 1, EOG: 2)`
;however, my EOG channels seem to be classified as bad for some reason, could that be the error?

agramfort, 2018-02-23:
Yes you need to empty the bads

aitorme, 2018-02-23:
yeah, emptying the bad list worked fine, but now I get the same error as in #4852, will close this issue since this particular problem is solved, thanks!

closed 2018-02-23 by aitorme