mne-tools/mne-python #4968: Example script exits before plot renders

moppymopperson, 2018-02-26:
The example script for plotting the full MNE solution appears to be exiting and closing the results plot before the plot can be rendered or examined. The window appears for just a moment and then gets closed without any error message.

At first I presumed this was a problem with my installation, since mayavi is notoriously hard to get working properly, but I've confirmed the same behavior using fresh installs of MNE in dedicated Anaconda environments on both Windows 7 and Ubuntu Artsy. The behavior is also the same under both python 2 and the experimental python 3 setup, which makes me think it may be more than a bad setup on my end.

This from a vanilla terminal by the way, not Spyder, IPython or Jupyter Notebook. I have not tested those. If I can figure out the issue I'll submit a PR. I'd prefer to be told my install is messed up though.

larsoner, 2018-02-26:
Run with `python -i` to enable "interactive mode". Otherwise scripts exit after completion. `matplotlib` does some nice things with blocking that prevent figures from closing but mayavi doesn't, so we are a bit stuck with this behavior.

moppymopperson, 2018-02-26:
I'm glad that's all it was, thanks for the quick feedback. 
Is the need to run in interactive mode documented somewhere, or it considered general knowledge?

closed 2018-02-26 by moppymopperson

agramfort, 2018-02-27:
well we all tend to work with interactive shells. Typically ipython.

I only start "python myscript.py" on a cluster for long computations.