mne-tools/mne-python #4975: ICA properties don't work properly with manual rejections

cbrnr, 2018-02-27:
I noticed that there seems to be an issue with the following workflow:

1. Manually create annotations to mark bad segments in `raw`
2. Run ICA (by default this uses only segments in `raw` that are **not** marked as bad)
3. Inspect the components via `ica.plot_components(raw)`
4. Click on an IC to inspect it in more detail - the resulting ICA properties window is empty and I get the following error message:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/matplotlib/cbook/__init__.py", line 388, in process
    proxy(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/matplotlib/cbook/__init__.py", line 228, in __call__
    return mtd(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/mne/viz/topomap.py", line 1009, in onclick
    ica.plot_properties(inst, picks=ic, show=True)
  File "/usr/local/lib/python3.6/site-packages/mne/preprocessing/ica.py", line 1435, in plot_properties
    figsize=figsize, show=show)
  File "/usr/local/lib/python3.6/site-packages/mne/viz/ica.py", line 232, in plot_ica_properties
    epochs_src = ica.get_sources(inst)
  File "/usr/local/lib/python3.6/site-packages/mne/preprocessing/ica.py", line 748, in get_sources
    sources = self._sources_as_epochs(inst, add_channels, False)
  File "/usr/local/lib/python3.6/site-packages/mne/preprocessing/ica.py", line 798, in _sources_as_epochs
    sources = self._transform_epochs(epochs, concatenate)
  File "/usr/local/lib/python3.6/site-packages/mne/preprocessing/ica.py", line 675, in _transform_epochs
    data = np.hstack(epochs.get_data()[:, picks])
  File "/usr/local/lib/python3.6/site-packages/numpy/core/shape_base.py", line 288, in hstack
    return _nx.concatenate(arrs, 1)
ValueError: need at least one array to concatenate
```

This is caused by an empty `Epochs` object, which is normally displayed in the upper right corner of the properties window. It is empty because the `raw` data gets segmented into 2s epochs, and all of these epochs coincide with a `BAD` annotation segment.

I've never fully understood the reason for the 2s epochs, but I guess this figure is useful to show a component's activity over time at a glance. If there are many bad segments, these 2s epochs can't be generated and thus the error.

Not sure what the best solution is. I guess it is OK that all information in the properties window is derived from the data that ICA was actually computed on (i.e. on data excluding all `BAD` segments in the default case) - it doesn't make sense to show the full `raw` data here. Maybe in such a case the epochs image should just be skipped? Any ideas?

larsoner, 2018-02-27:
> Maybe in such a case the epochs image should just be skipped? Any ideas?

That sounds fine for now, probably with a `warn` about it. If you or someone else in a similar situation can think of a way to summarize such broken-up data then we can always add it back later.

agramfort, 2018-02-27:
sounds good to me

mmagnuski, 2018-02-28:
Another option would be to ignore `BAD` annotations in such case. Or always ignore `BAD` annotations when using raw and only mark BAD regions somehow in the plot (NaNs for example, but that would break smoothing, so might be better to use a transparency layer...). But I agree that the easiest short term solution is to ignore the epochs plot.
2s windows are used to keep it similar to eeglab (which uses 1s windows with raw IIRC).

cbrnr, 2018-02-28:
Ignoring `BAD` annotations is not a good idea, because then the plot shows data that was not used in the ICA calculation. I like the idea to use some kind of filler values which translate to white in the plot.