mne-tools/mne-python #4979: BUG:  permutation clustering 2samp test depends on data order?

dengemann, 2018-02-28:


britta-wstnr, 2018-03-05:
Hi @dengemann - just came across this,
isn't this due to the fact that the indices for the permutations are the same due to the seed, however, they index different trials since X is reversed - meaning your permuted data sets will look different?
I took a quick look and for me, the two solutions converge when using more permutations (`n_permutations=5000`).

dengemann, 2018-03-05:
> Hi @dengemann - just came across this,

Hi Britta, thanks for anticipating the right test. I independently concluded that it must be arbitrary initial differences that get randomly amplified by the seed but should converge at some point. Great to hear that it seems to be this. This adds to the discussion in #4953 -- and reinforces me in thinking that we want to have a small tutorial somewhere that shows how the impact of seeds diminishes with numbers of permutations.

larsoner, 2018-04-18:
Closing as dup of #4953

closed 2018-04-18 by larsoner