mne-tools/mne-python #4982: mne anonymize command

agramfort, 2018-03-03:
we should have the command equivalent of

```
$ mne_anonymize
usage: mne_anonymize [options]
Remove subject information from a fif file.
	--his             remove the HIS ID tag as well
	--file name       the file to modify.
	--help            print this info.
	--version         print version info.
```

dokato, 2018-03-12:
I'm happy to take it, if no one started.

agramfort, 2018-03-12:
go for it.

it would be nice to allow be able to operate inplace too but fine to do it
in a subsequent PR.

dokato, 2018-03-12:
okay, but please help me understand `--his` parameter.
Since in `mne.io.meas_info.anonymize_info` we remove whole `subject_info` key so `his_id` as well by default. Should I modify `anonymize_info` function to preserve this depending on some `his` flag?

agramfort, 2018-03-12:
removing it all the time is good. Forget the --his param.

drammock, 2020-11-17:
this was closed in #6892

closed 2020-11-17 by drammock