mne-tools/mne-python #4984: problem with reading fieldtrip data

dokato, 2018-03-05:
I was trying to load a dataset from fieldtrip and noticed a small bug, when trying to plot the signal:

```
/mne/viz/utils.pyc in _handle_decim(info, decim, lowpass)
   2210         lp = min(lp, info['sfreq'] if lowpass is None else lowpass)
   2211         info['lowpass'] = lp
-> 2212         decim = max(int(info['sfreq'] / (lp * 3) + 1e-6), 1)
   2213     decim = _ensure_int(decim, 'decim', must_be='an int or "auto"')
   2214     if decim <= 0:

ValueError: cannot convert float NaN to integer
```

It's because:
```
highpass : float | nan Hz
lowpass : float | nan Hz
```
in `raw.info` and MNE assumes that it will be either number or None.

To reproduce this bug you can follow sections _Procedure > Export to Raw_ from this tutorial:
http://www.fieldtriptoolbox.org/development/integrate_with_mne
 and call eg. `raw.plot()`.

I can help with fixing it, but wonder if we should check for NaN in this function, or rather while loading the data and making info?

agramfort, 2018-03-05:
please send a PR

should it be fixed in Fieldtrip?

dokato, 2018-03-05:
@agramfort In case of not detecting that info in FT header it assigns default values as (funct `fieldtrip2fiff.m`):
```
  info.highpass = NaN;
  info.lowpass  = NaN;
```
I think we can just replace all nans by Nones, since MNE can deal with it later on.

agramfort, 2018-03-05:
yes sounds good

closed 2018-03-06 by dokato