mne-tools/mne-python #4987: Make `epochs._hid_match` public?

jona-sassenhagen, 2018-03-05:
It's straight-forward to go from epochs with HIDs to dicts of evokeds with HIDs, e.g., 

```
evokeds = {cond: epochs[cond].average() for cond in epochs.event_id}

(result)
evokeds = {
    "aud/left": evoked, "aud/right": evoked,
    "vis/left": evoked, "vis/right": evoked}
```

Now it would be cool to do factorial contrasts with, e.g., in a perfect world,

```
mne.combine_evokeds((evokeds["aud"], evokeds["vis"]), weights=(-.5, .5)).plot_joint("aud minus vis")
```

We can't quite go there, but this would be facilitated by making `epochs._hid_match` public, for

```
evokeds_ = (mne.grand_average(evokeds[key] for key in  hid_match(cond, evokeds))
            for cond in ("aud", "vis"))
mne.combine_evokeds(evokeds_, weights=(-.5, .5)).plot_joint("aud minus vis")
```

And perhaps even

```
evokeds_ = [hid_match(evokeds, key) for key in ("aud", "vis")]
mne.combine_evokeds(evokeds_, weights=(-.5, .5)).plot_joint("aud minus vis")
```

agramfort, 2018-03-05:
to do this I would write my own container an Evoked that understands `
evokeds["aud"]`
you would not need any change to MNE then I think

jona-sassenhagen, 2018-03-05:
That's true :) So I close?

agramfort, 2018-03-05:
I would say so :)

jona-sassenhagen, 2018-03-05:
:(

closed 2018-03-05 by jona-sassenhagen