mne-tools/mne-python #4989: autoreject (global) in MNE

jasmainak, 2018-03-05:
Since I have a little "free time" these days, I am wondering if there is some interest in adding `autoreject` to MNE. Although the API of autoreject (local) is not yet settled, I think adding autoreject (global) would be quite easy without any additional dependencies. We could do something like:

```py
import mne
mne.Epochs(..., reject="autoreject",)
ICA(..., reject="autoreject")
```

In the `autoreject` repository, I can internally call the MNE function so that those relying on it are not affected. Let me know.

dengemann, 2018-03-06:
+1 after doing the 0.1 release :)

dengemann, 2018-03-06:
I'd probably not bake this into the epochs constructor though as, usually, you interact with the autoreject outputs. Where the hack does the aureject object go Mainak when you do it the way you suggest here?

jasmainak, 2018-03-06:
I'm actually suggesting something more simple than what you are probably thinking. What I propose is to do it in small steps. So to start with, I would just move the functionality offered by [this example](http://autoreject.github.io/auto_examples/plot_estimate_global_reject.html#sphx-glr-auto-examples-plot-estimate-global-reject-py). And of course, it wouldn't be baked into the Epochs object. It would just be one line extra to call this method inside the constructor.

dengemann, 2018-03-06:
I am thinking at this point we should just copy the future autoreject API
into MNE preprocessing.
Then we can copy over some of the examples.
Then eventually add reject=‘auto’ when we know how to then get fixlog /
droplog etc.
As long as you want to look at autoreject results you want to have the
object. That’s ok. Also keep in mind how reject interacts with .get_data
and preloading. You don’t want to wait for autoreject until you get your
data from disk.

agramfort, 2018-03-06:
what is preventing a 0.1 release now?

dengemann, 2018-03-06:
Api changes that are only 80 % realized and missing tests

jasmainak, 2018-03-06:
okay, let's discuss it in Inria next week :-)

jona-sassenhagen, 2018-03-06:
I like autoreject. If we're adding cutting-edge ICA like Picard, may as well add autoreject.

dengemann, 2018-03-06:
It's happening Jona! But not before a proper release on pip and finishing
the API transition. Good news: we will do it these days.

jona-sassenhagen, 2019-03-24:
This issue is still open!!! @jasmainak @dengemann

jasmainak, 2019-03-24:
okay for me to close it. We can keep maintaining autoreject in a separate repo for now.

jona-sassenhagen, 2019-03-24:
Ok. Would still be cool to have it here, it's a great tool!

closed 2019-03-24 by jona-sassenhagen