mne-tools/mne-python #4991: applying projectors to CTF data (BUG?)

bloyl, 2018-03-06:
I'm seeing strange behavior applying EOG and ECG projectors to CTF datasets.

with epochs created with `proj='delayed' I'm looking at the effect of the projection operators using the following code.
```
times = 1e-3 * np.array([86, 148, 94, 160])
epochs.average().plot_topomap(times, title='No Proj', proj=False,
                              vmin=-100, vmax=100, show=False)
epochs.average().plot_topomap(times, title='Proj', proj=True,
                              vmin=-100, vmax=100, show=False)
epochs.average().plot_topomap(times, title='Delayed Proj',
                              proj='interactive',
                              vmin=-100, vmax=100, show=True)
```

I have 2 issues. 
1) with all the checkboxes checked the topoplots don't match the maps from `proj=True`, I like the interactive plots better.
2) the check boxes aren't reversible. if I click all of them and then unclick them all I don't end up where I started.

you can find screen shots here the show this 
https://www.dropbox.com/sh/dqdirsnre6hqma8/AAByY4MuxUbBozcKsvFAJ4Qya?dl=0

apply_proj_nothingChecked.png : the delayed plot matched the No Proj plot
apply_proj_allChecked.png: the delayed plot does not match the Proj plot. 
apply_proj_nothingChecked2.png: this is the delayed plot with nothing checked after selecting everything. it does not match the No Proj plot. 

Am I just doing something wrong? The code for applying the projs interactively (https://github.com/mne-tools/mne-python/blob/master/mne/viz/topomap.py#L100-L104) seems like it should match `proj=True` case and be fully reversible.  So i'm abit baffled. Does any one have any ideas?

larsoner, 2018-03-06:
> Am I just doing something wrong? The code for applying the projs interactively (https://github.com/mne-tools/mne-python/blob/master/mne/viz/topomap.py#L100-L104) seems like it should match proj=True case and be fully reversible.

Yes this is supposed to be the behavior. We must have introduced a regression somewhere that is causing this. I could try to address this in:

https://github.com/mne-tools/mne-python/pull/4972

As it aims to unify `raw` and `epochs` plotting, which is two places where this can happen. But there are probably more. It's probably worth doing a separate, more manageable PR to fix only the bug(s) reported here.

larsoner, 2018-04-03:
@bloyl can you see if this behaves as you'd expect:
```
import mne
data_path = mne.datasets.sample.data_path()
raw = mne.io.read_raw_fif(data_path + '/MEG/sample/sample_audvis_raw.fif')
raw.load_data().pick_types(meg='mag', stim=True)
events = mne.find_events(raw, 'STI 014')
epochs = mne.Epochs(raw, events, proj='delayed')
evoked = epochs.average()
kwargs = dict(vmin=-100, vmax=100)
evoked.plot_topomap(proj=False, title='No proj', **kwargs)
evoked.plot_topomap(proj=True, title='Proj', **kwargs)
evoked.plot_topomap(proj='interactive', title='Interactive', **kwargs)
```
If so, then it could be a problem with CTF compensation channels or something.

If not, then I'll be confused because it seems to work fine over here.

larsoner, 2018-04-03:
Just latest `master` is fine. That's more or less where I was testing it.

Looking closer I agree, the contours at least are broken / don't update. I'll try to look into it soon to get it into 0.16.