mne-tools/mne-python #4992: BUG: subject morph

kingjr, 2018-03-06:
We're getting a weird dimension mismatch when trying to morph some source estimate after a beamformer.

I dont really know what to do..

```python
import mne
stc = mne.read_source_estimate('test-stc.fif')
stc.subject = 'MRS02' # for some reason, stc.save didnt save the name of the subject, which create another bug...
stc.morph('fsaverage', subjects_dir='/Applications/freesurfer/subjects/')
```

stc files
https://ufile.io/kdnvu
https://ufile.io/o8exl

error:
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-7-ccbf4a676c3e> in <module>()
----> 1 stc.morph('fsaverage', subjects_dir='/Applications/freesurfer/subjects')

/Users/poeppellab/Applications/mne-python/mne/source_estimate.pyc in morph(self, subject_to, grade, smooth, subjects_dir, buffer_size, n_jobs, subject_from, sparse, verbose)

/Users/poeppellab/Applications/mne-python/mne/utils.pyc in verbose(function, *args, **kwargs)
    726         with use_log_level(verbose_level):
    727             return function(*args, **kwargs)
--> 728     return function(*args, **kwargs)
    729 
    730 

/Users/poeppellab/Applications/mne-python/mne/source_estimate.pyc in morph(self, subject_to, grade, smooth, subjects_dir, buffer_size, n_jobs, subject_from, sparse, verbose)
   1315         else:
   1316             return morph_data(subject_from, subject_to, self, grade, smooth,
-> 1317                               subjects_dir, buffer_size, n_jobs, verbose)
   1318 
   1319     def morph_precomputed(self, subject_to, vertices_to, morph_mat,

/Users/poeppellab/Applications/mne-python/mne/source_estimate.pyc in morph_data(subject_from, subject_to, stc_from, grade, smooth, subjects_dir, buffer_size, n_jobs, warn, verbose)

/Users/poeppellab/Applications/mne-python/mne/utils.pyc in verbose(function, *args, **kwargs)
    726         with use_log_level(verbose_level):
    727             return function(*args, **kwargs)
--> 728     return function(*args, **kwargs)
    729 
    730 

/Users/poeppellab/Applications/mne-python/mne/source_estimate.pyc in morph_data(subject_from, subject_to, stc_from, grade, smooth, subjects_dir, buffer_size, n_jobs, warn, verbose)
   2416                                      warn=warn)
   2417                      for data_buffer
-> 2418                      in np.array_split(data[hemi], n_chunks, axis=1)), axis=1)
   2419 
   2420     vertices = [nearest[0], nearest[1]]

/Users/poeppellab/Applications/mne-python/mne/source_estimate.pyc in <genexpr>((data_buffer,))
   2415                                      n_vertices, nearest[hemi], maps[hemi],
   2416                                      warn=warn)
-> 2417                      for data_buffer
   2418                      in np.array_split(data[hemi], n_chunks, axis=1)), axis=1)
   2419 

/Users/poeppellab/Applications/mne-python/mne/source_estimate.pyc in _morph_buffer(data, idx_use, e, smooth, n_vertices, nearest, maps, warn, verbose)

/Users/poeppellab/Applications/mne-python/mne/utils.pyc in verbose(function, *args, **kwargs)
    726         with use_log_level(verbose_level):
    727             return function(*args, **kwargs)
--> 728     return function(*args, **kwargs)
    729 
    730 

/Users/poeppellab/Applications/mne-python/mne/source_estimate.pyc in _morph_buffer(data, idx_use, e, smooth, n_vertices, nearest, maps, warn, verbose)
   2251     logger.info('    %d smooth iterations done.' % (k + 1))
   2252 
-> 2253     data_morphed = maps[nearest, :] * data
   2254     return data_morphed
   2255 

//anaconda/lib/python2.7/site-packages/scipy/sparse/base.pyc in __mul__(self, other)
    474 
    475             if other.shape[0] != self.shape[1]:
--> 476                 raise ValueError('dimension mismatch')
    477 
    478             result = self._mul_multivector(np.asarray(other))

ValueError: dimension mismatch
```

agramfort, 2018-03-08:
but morph should create the morph-maps so it should work as is no?

larsoner, 2018-03-08:
> but morph should create the morph-maps so it should work as is no?

Only if we have the proper Freesurfer data for the subject, which I was trying to avoid sending.

kingjr, 2018-03-12:
(this is @avkasdan)

https://ufile.io/x5lsp - morph map for this subject
https://ufile.io/7v4ru - forward model
https://ufile.io/9v02e - data covariance
https://ufile.io/sgxmn - noise covariance 
https://ufile.io/hqsl5 - script which throws the error and makes STCs

Thank you!!!

agramfort, 2018-03-12:
can you just share a dropbox folder to it's painless to replicate the pb?

thx

avkasdan, 2018-03-12:
https://www.dropbox.com/sh/wl8a6s2mp25u30e/AAA9rcqYgMeEwCSGPmgfAO6Ba?dl=0 

Here is the dropbox link, let me know if that works

avkasdan, 2018-03-14:
Any chance you have had a look at this yet?

agramfort, 2018-03-14:
the code morphing.py you shared has many hard coded path so there is no
easy way for me to replicate the error you report.

I will have a look if running morphing.py also me to replicate the error
easily.

thanks

kingjr, 2018-03-14:
Anna I cleaned up your code and it runs ok on my machine. 

```python
import os
import mne
from mne.cov import compute_covariance
import numpy as np

subject = 'MRS02'
fwd_fname = '%s-fwd.fif' % subject
epo_fname = '%s_audio_None-30.0Hz-epo.fif' % subject
morph_fname = 'fsaverage-%s-morph.fif' % subject
subjects_dir = './'

fwd = mne.read_forward_solution(fwd_fname)

# Read in events and epochs, and apply baseline
epochs = mne.read_epochs(epo_fname, verbose=False)
epochs.apply_baseline((-1.300, -1.0))

# Here is where we ignore the ref channels (can't do before, fwd complains)
epochs.pick_types(meg=True, ref_meg=False)

noise_cov = compute_covariance(epochs, tmin=-1.300, tmax=-1.)
data_cov = compute_covariance(epochs, tmin=.000, tmax=.650)

# Compute beamforming solution
filters = mne.beamformer.make_lcmv(epochs.info, fwd, noise_cov=noise_cov,
                                   reg=0.01, data_cov=data_cov,
                                   verbose=False, pick_ori='max-power')

# average evoked response
evoked = epochs.average()

# Apply beamforming solution
stc = mne.beamformer.apply_lcmv(evoked, filters)

# apply morph
fs_vertices = [np.arange(163842)] * 2
vertices_to = [np.arange(163842)] * 2
morph = mne.compute_morph_matrix(subject, 'fsaverage',
                                 vertices_from=stc.vertices,
                                 vertices_to=fs_vertices,
                                 subjects_dir=subjects_dir)
stc_morph = mne.morph_data_precomputed(subject, 'fsaverage', stc,
                                       vertices_to, morph)
```

Can you check that the above work and identified what is the step in your script that creates the error?

PS: [here](https://wetransfer.com/downloads/ac95315200e373d63505187f9b83ba9820180314213005/e7813587c2e8cf10e77f03b378c9a90720180314213005/6f9e6f) are all the files necessary to run the above script.

avkasdan, 2018-03-14:
The above works until the mne.compute_morph_matrix line where I get the error- a dimension mismatch.

kingjr, 2018-03-15:
try deleting the morph file in the fsaverage folder, the morph function
will recreate it

avkasdan, 2018-03-15:
this is seeming to work!! Thanks!

avkasdan, 2018-03-15:
I am going to close this issue- I think the morph maps I was using were older ones, and it worked to compute them here on the fly.

closed 2018-03-15 by kingjr