mne-tools/mne-python #4998: Bug in IO module for BrainVision: Wrong conversion from Seconds to Hertz

sappelhoff, 2018-03-07:
In BrainVision Recorder *.vhdr files, the highpass filter settings are often listed as "low-cutoff in seconds". The relevant mne.io function translates this into Hz as such:

https://github.com/mne-tools/mne-python/blob/239c83dfe0686084d1c96c945579dc8cc10e895c/mne/io/brainvision/brainvision.py#L597-L598

With this method, the standard setting of `low-cutoff=10s` is translated to highpass=0.1Hz. However, the [BrainProducts website](http://www.brainproducts.com/productdetails.php?id=2) specifies 0.016Hz (see under "Lower cutoff frequency (high pass) / time constant").

The conversion from seconds to Hz is probably not implemented correctly in the mne.io function for brainvision. Or am I missing something?

Some references:
- [book excerpt with brief explanation and equation](https://books.google.se/books?id=z-PmHHsi4zYC&pg=PA41&lpg=PA41&dq=cut+off+frequency+conversion+hertz+and+seconds&source=bl&ots=g8Cf68JVOy&sig=CuDjd5p59iyyemfpuHu8AUlpnc0&hl=en&sa=X&ved=0ahUKEwi10t-z2trZAhVSsaQKHWh-DS84ChDoAQgmMAA#v=onepage&q=cut%20off%20frequency%20conversion%20hertz%20and%20seconds&f=false)

- [handy calculator and equations](http://www.sengpielaudio.com/calculator-timeconstant.htm)

sappelhoff, 2018-03-07:
The fix seems simple enough:

``` python
if hp_s:
    info['highpass'] = 1. / (2 * np.pi * info['highpass'])
```

... and using this approach also for the remaining lines (lowpass filters, and the case for heterogeneous filters)

larsoner, 2018-03-07:
Yes this looks correct to me. I've actually never seen this nomenclature before.

larsoner, 2018-03-07:
... although "time constant" rings a bell in terms of analog filtering and circuits. I wonder if that's where it comes from.

sappelhoff, 2018-03-07:
I find it unnecessarily confusing and I am wondering about the benefits of specifying 10seconds instead of 0.016Hz ... maybe because it is actually `0.015915494309189534` and then 10seconds is more straight forward and doesn't need rounding? Anyways, I have opened a PR

dmedine, 2018-03-09:
Hello. My name is David Medine and I am a developer at Brain Products. The reason they use a time constant to represent the nature of a filter is for strange, convoluted, historical reasons. 

What is important is that sappelhoff is correct. The denominator should have a factor of 2pi in it.

larsoner, 2018-03-09:
Thanks for the confirmation @dmedine. Looks like #5000 should be correct, then.

PR #5000 (2018-03-09): fixes #4998
 fixes #4998 

Regarding the conversion from seconds to Hertz when reading filter information from a BrainVision Recorder *.vhdr file.
commit 58e59df8 (2018-03-09): fixes #4998 (#5000)

* fixes #4998

*  add comments and reference for brainvision s to Hz.

* updated tests for brainvision io

* pep8 fixes and add author brainvision io

* added changes to log
closed 2018-03-09 by larsoner