mne-tools/mne-python #5005: Images missing on the tutorial website

jona-sassenhagen, 2018-03-10:
E.g. the ones at the end here:

https://mne-tools.github.io/stable/manual/decoding.html

@larsoner

larsoner, 2018-03-12:
This should be fixed by turning this `manual` entry into a `tutorials` script. It should be easy for someone to do when they are bored

cbrnr, 2018-03-30:
But which parts stay in the manual and which parts will be moved to a new tutorial script?

larsoner, 2018-03-30:
> But which parts stay in the manual and which parts will be moved to a new tutorial script?

Eventually I think all of `manual` should / could be in `tutorials`.

jona-sassenhagen, 2018-04-02:
[image]

This is what I see ...

larsoner, 2018-04-02:
Argh yes, temporary blindness on my part

PR #5076 (2018-04-02): MRG: Doc fixes
Fixes a bunch of little problems with our docs.

Closes #5005.
closed 2018-04-02 by larsoner