mne-tools/mne-python #5013: bad channels crash make_forward_solution

dnacombo, 2018-03-13:
When running mne.make_forward_solution with a CTF dataset, we get the following error message:
```
  File "/home/mchaumon/anaconda2/lib/python2.7/site-packages/mne/forward/_compute_forward.py", line 807, in _compute_forwards_meeg
    both[:, picks] = B

ValueError: shape mismatch: value array of shape (22494,150) could not be broadcast to indexing result of shape (149,22494)

```
We understood it's a fairly straightforward problem here lines 800 and following in _compute_forward.py :
```
        # Compensate if needed (only done for MEG systems w/compensation)
        if compensator is not None:
            # Compute the field in the compensation sensors
            work = fun(rr, mri_rr, mri_Q, ccoils, csolution, bem_rr,
                       n_jobs, coil_type)
            # Combine solutions so we can do the compensation
            both = np.zeros((work.shape[0], B.shape[1] + work.shape[1]))
            picks = pick_types(info, meg=True, ref_meg=False)
            both[:, picks] = B
            picks = pick_types(info, meg=False, ref_meg=True)
            both[:, picks] = work
            B = np.dot(both, compensator.T)
```
where bad channels need to be sliced away of `B`, but indices `picks` picked in the info dictionary containing all channels don't map into array B that has only MEG channels. Don't know how to fix that properly myself, but I suspect it's rather straightforward to fix...

Note this does not reproduce with the sample dataset, for which `compensator` is `None` below...

Keep @jdammers posted.

agramfort, 2018-03-13:
can you share some data to replicate the pb?

dnacombo, 2018-03-13:
https://owncloud.icm-institute.org/index.php/s/IFRyr4O5c16JUk8

agramfort, 2018-03-13:
please share a full folder with a script to replicate easily the bug.

thanks

dnacombo, 2018-03-13:
Sorry: https://owncloud.icm-institute.org/index.php/s/jxBtcntc1MXBVAt
Thanks!

agramfort, 2018-03-13:
see https://github.com/mne-tools/mne-python/pull/5016 for the fix.

thanks a lot for the bug report.

PR #5016 (2018-03-14): Fix comp fwd ctf bads
closes #5013
closed 2018-03-14 by larsoner