mne-tools/mne-python #5015: BUG: CircleCI 1.0 is retiring

larsoner, 2018-03-13:
We need to move to 2.0 by August.

We should also switch to using `environment.yml`, which means having a newer version of Mesa (the one of 14.04 is too old to support the necessary OpenGL extensions for VTK7+).

agramfort, 2018-03-24:
here is an example of the new API we could reuse:

https://github.com/NeuralEnsemble/python-neo/tree/master/.circleci

larsoner, 2018-03-24:
SciPy also just transitioned. The annoying part for us will be getting a
usable docker image with OSMesa for VTK rendering.

agramfort, 2018-03-24:
see if you find any repo on github that uses vtk with circleci

larsoner, 2018-03-25:
Maybe this one

https://github.com/conda-forge/pyvtk-feedstock

I'll try to look this week