mne-tools/mne-python #5018: *TFR obj data attribute documentation?

jona-sassenhagen, 2018-03-14:
The docs for the TFR objects do not specify how to get at the data. Should this not be included in the doctoring?

Colleagues of mine have complained to me MNE is "sparsely documented" here :)

@agramfort @palday

agramfort, 2018-03-14:
on an AverageTFR ?
https://martinos.org/mne/dev/generated/mne.time_frequency.AverageTFR.html

yes a good docstring would help and also we should document attributes not
just parameters

jona-sassenhagen, 2018-03-14:
Yes, and https://martinos.org/mne/dev/generated/mne.time_frequency.EpochsTFR.html#mne.time_frequency.EpochsTFR

I can go about it once the other PRs are merged :)

agramfort, 2018-03-14:
ok convince a less experienced person to do this change as it's only a
documentation update.

jona-sassenhagen, 2018-03-14:
@nbara do you want to do it? Or is it not enough of a challenge? :)

nbara, 2018-03-15:
@jona-sassenhagen it's not about the challenge, but I'm afraid my hands are full at the moment, sorry... 
I was giving a thumbs up because a colleague whom I've slowly managed to convince to try MNE has been complaining about the same thing :)

jona-sassenhagen, 2018-03-23:
@sappelhoff maybe you wanna try your hands at this one though?

sappelhoff, 2018-03-23:
@jona-sassenhagen, I haven't worked with time frequency data in MNE yet. I wouldn't really know where to start looking up information that should go into the docstring ...

jona-sassenhagen, 2018-03-23:
I would say that it should minimally just include that AverageTFR and EpochsTFR have an attribute (data) that includes the data, and the format (which depends on which kind of object it is). There is an Attributes section in the docs, but it does not include that info.

sappelhoff, 2018-03-26:
So for the `EpochsTRF` first ... given the code below, you would want a description of 
>'ch_names', 'comment', 'compensation_grade', 'data', 'freqs', 'info', 'method', 'times'

in the docstring, where currently only `'ch_names'` is implemented?

``` python 
In [1]: import mne
In [5]: import numpy as np

In [10]: data = np.random.random(100, 64, 100, 1000)
In [11]: times = np.arange(1000)
In [12]: freqs = np.linspace(1,100,100)
In [16]: info = mne.create_info(32,1000)

In [17]: mne.time_frequency.EpochsTFR(info, data, times, freqs)

In [18]: epochsTFR = mne.time_frequency.EpochsTFR(info, data, times, freqs)

In [19]: dir(epochsTFR)
Out[19]: 
['__abs__',
 '__class__',
 '__contains__',
 '__delattr__',
 '__dict__',
 '__doc__',
 '__format__',
 '__getattribute__',
 '__hash__',
 '__init__',
 '__module__',
 '__new__',
 '__reduce__',
 '__reduce_ex__',
 '__repr__',
 '__setattr__',
 '__sizeof__',
 '__str__',
 '__subclasshook__',
 '__weakref__',
 '_data',
 '_pick_drop_channels',
 '_size',
 'add_channels',
 'apply_baseline',
 'average',
 'ch_names',
 'comment',
 'compensation_grade',
 'copy',
 'crop',
 'data',
 'drop_channels',
 'freqs',
 'info',
 'method',
 'pick_channels',
 'pick_types',
 'preload',
 'save',
 'times']
```

jona-sassenhagen, 2018-03-26:
First and foremost `data`, but if you can do a few others, that'd be good too. Refer to the Evoked and Epochs objects. Does that work for you?

sappelhoff, 2018-03-29:
yep, looking at the other objects helped. After that it was just copy pasting and little adjustments e.g., for the data shape. see the PR.

commit b1448c0b (2018-03-29): DOC: add attribute description to epochs and average TFT, fixes #5018 (#5056)
closed 2018-03-29 by agramfort