mne-tools/mne-python #5022: problem with mne.decoding.Scaler

giuliagennari, 2018-03-15:
I am trying to standardize my epochs:
from mne.decoding import Scaler
epochs_data = epochs.get_data()
epochs_sc = Scaler().fit_transform(epochs_data)

And I get this error:
[image]

As if an epochs object and a numpy array were required at the same time.
Could anybody help me out?

kingjr, 2018-03-15:
Hi,

By default the Scaler objects scale magnetometers, gradiometers and EEG
channels by a fixed amount: see details on the documentations
https://martinos.org/mne/dev/generated/mne.decoding.Scaler.html

This means that you need to inform the object about the channel
identify:e.g.
Scaler(info=epochs.info).fit_transform(epochs.get_data())

Else, you can scale the data using a variety of statistical approach: e.g.
Scaler(scalings='mean').fit_transform(epochs.get_data())

Best regards,

Jean-Remi

giuliagennari, 2018-03-15:
That's right, I completely forgot to specify scalings='mean'.

Thank you very very much Jean-Remi!!!

Best,

Giulia

jona-sassenhagen, 2018-03-23:
The error message is a bit cryptic ... we should probably do a better one (`Both "info" and "scalings" can't be None, please specify` etc).

jona-sassenhagen, 2018-03-25:
@sappelhoff this would also be an easy thing to do, if you're still looking :)

sappelhoff, 2018-03-25:
haha @jona-sassenhagen - thanks for finding some easy tasks for me ;-) 

something like this ... #5042 ?

PR #5042 (2018-03-26): MINOR: Specify err msg scaler
ENH: better error msg in decoding.scaler. Closes #5022
commit ce59305d (2018-03-26): MINOR: Specify err msg scaler (#5042)

* DOC: Add information about used units, fixes #5003

* ENH: better error msg in decoding.scaler. Closes #5022

* added test

* put err msg at more sensible line and change err msg content

* use % string formatting
closed 2018-03-26 by larsoner