mne-tools/mne-python #5024: How to know which event_id corresponds to which event. BCI comptetion 3 (Dataset 3a)

vivek9746, 2018-03-16:
The data description stated 5 classes 1,2,3,4,NaN, corresponding to the left hand, right hand, foot and tongue. whereas I have got a different result on executing:

>>>events = mne.find_events(raw,shortest_event=0, stim_channel='STI 014')

1080 events found
Events id: [ 768 1791 2323 2324 2325 2326 2337 2339 3346 3347 3348 3349 3360 3362]

Can anyone tell me what is this data and how do I extract information related to the classes in them

teonbrooks, 2018-03-16:
hey @vivek9746, it's hard to tell from the description alone. do you have a copy of the file that you share publicly so we can take a look or a snippet of it?

agramfort, 2018-03-16:
probably @cbrnr or @alexandrebarachant can help

cbrnr, 2018-03-16:
Without the file it's hard to tell which events are actually used in the file. Can you share the file? If it is a GDF file, you might want to try `raw.find_edf_events()` - this will give you the events directly as stored in the file (as opposed to `mne.find_events`, which converts these events to a stim channel). I know that event 768 (hex 0x300) is used for trial start in BioSig (see [here](https://github.com/donnchadh/biosig/blob/master/biosig/doc/eventcodes.txt) for a list of known event codes), so the remaining codes you get could either be custom codes or a result from a parsing problem.

vivek9746, 2018-03-16:
I have used the Dataset 3(a) of BCI Competition 3. I started working on data subject "k3_b". The dataset and its description can be found on the following link:  https://drive.google.com/open?id=15PFMvttxvy3SfKXOsELI1SjmjY9UHFXx

cbrnr, 2018-03-16:
You can use `raw.find_edf_events()` to get all 1502 events, e.g. like this:

```
raw = mne.io.read_raw_edf("k3b.gdf", preload=True)
_, pos, kind, chan, dur = raw.find_edf_events()
```

The event numbers in `kind` are decimal event codes, you need to convert them to hex to find the in the [event code list](https://github.com/donnchadh/biosig/blob/master/biosig/doc/eventcodes.txt):

```
kind_hex = np.array([hex(n) for n in kind])
```

You will find that these events are present in the file:

```
np.unique(kind_hex)
```

```
array(['0x300', '0x301', '0x302', '0x303', '0x304', '0x30f', '0x311',
       '0x312', '0x3ff'], dtype='<U5')
```

`0x300` is trial start, and `0x301`-`0x304` correspond to classes 1-4.

vivek9746, 2018-03-19:
Thank you for the help @cbrnr. :)

closed 2018-03-19 by vivek9746