mne-tools/mne-python #5025: Bug in peak_finder() and find_eog_events()

HariSeldon11988, 2018-03-16:
If no peaks were found there is _TypeError: can't multiply sequence by non-int of type 'float'_,
because in line 161 of peak_finder.py an empty list is multiplied by -1.

A solution is to put in lines 156 and 157:
```
            peak_mags = np.array([])
            peak_inds = np.array([])
```
instead of
```
            peak_mags = []
            peak_inds = []
```

However, it will only delay a problem. Full solution has to contain also a verification of eog_events length in 68th line of eog.py, e.g.:
```
    if len(eog_events) > 0:
        eog_events[:, 0] = np.round(times[eog_events[:, 0] -
                                      raw.first_samp]).astype(int)
```
instead of just:
```
    eog_events[:, 0] = np.round(times[eog_events[:, 0] -
                                      raw.first_samp]).astype(int)
```
which assumes that eog_events is not empty.

agramfort, 2018-03-17:
please share a code snipped to replicate the pb. It will be used as
non-regression test.

thanks

mmagnuski, 2018-03-20:
I've had a similar problem before using bare `peak_finder` (`ValueError: zero-size array to reduction operation maximum which has no identity`), here is how to replicate:
```python
import numpy as np
from mne.preprocessing.peak_finder import peak_finder

values = np.arange(2, 1, 0.05)
peak_finder(values)
```
maybe just fixing `peak_finder` would be enough?

agramfort, 2018-03-20:
yes fix welcome

TommyClausner, 2018-03-25:
Hey there! Guess I would like to fix it if not already done

agramfort, 2018-03-25:
Go for it. Thx

PR #5040 (2018-03-26): FIX: Bug in peak_finder() and find_eog_events() #5025
Fixes #5025 
Problems resulted mostly from data type handling. I added  two
statements checking whether the output type is as desired (i.e. ndarray
of dtype int64 or float64). Further, problems raised, when the input
vector was empty. Added a check to the check for dimensionality.
closed 2018-03-26 by agramfort