mne-tools/mne-python #5026: Hash mismatch error

krishamehta, 2018-03-17:
When I try to download the spm dataset, this is the error I get.

`Automatically created module for IPython interactive environment
Local file appears to be complete (file_size == initial_size == 924932163)
<ipython-input-64-9f3225ed7000>:11: RuntimeWarning: Local file appears to be complete (file_size == initial_size == 924932163)
  data_path = spm_face.data_path()
Error while fetching file https://files.osf.io/v1/resources/rxvq7/providers/osfstorage/59c0e343b83f6902538c4553?action=download&version=2&direct. Dataset fetching aborted.
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-64-9f3225ed7000> in <module>()
      9 print(__doc__)
     10 
---> 11 data_path = spm_face.data_path()
     12 subjects_dir = data_path + '/subjects'

~/Documents/Projects/MachineLearning/libraries/lib/python3.5/site-packages/mne/datasets/spm_face/spm_data.py in data_path(path, force_update, update_path, download, verbose)

~/Documents/Projects/MachineLearning/libraries/lib/python3.5/site-packages/mne/utils.py in verbose(function, *args, **kwargs)
    726         with use_log_level(verbose_level):
    727             return function(*args, **kwargs)
--> 728     return function(*args, **kwargs)
    729 
    730 

~/Documents/Projects/MachineLearning/libraries/lib/python3.5/site-packages/mne/datasets/spm_face/spm_data.py in data_path(path, force_update, update_path, download, verbose)
     20     return _data_path(path=path, force_update=force_update,
     21                       update_path=update_path, name='spm',
---> 22                       download=download)
     23 
     24 data_path.__doc__ = _data_path_doc.format(name='spm',

~/Documents/Projects/MachineLearning/libraries/lib/python3.5/site-packages/mne/datasets/utils.py in _data_path(path, force_update, update_path, download, name, check_version, return_version, archive_name)
    374         full_name = list()
    375         for u, an, h, fo in zip(url, archive_name, hash_, folder_orig):
--> 376             remove_archive, full = _download(path, u, an, h)
    377             full_name.append(full)
    378         del archive_name

~/Documents/Projects/MachineLearning/libraries/lib/python3.5/site-packages/mne/datasets/utils.py in _download(path, url, archive_name, hash_)
    433             logger.info('Downloading archive %s to %s' % (archive_name, path))
    434             _fetch_file(url, full_name, print_destination=False,
--> 435                         hash_=hash_)
    436     return remove_archive, full_name
    437 

~/Documents/Projects/MachineLearning/libraries/lib/python3.5/site-packages/mne/utils.py in _fetch_file(url, file_name, print_destination, resume, hash_, timeout, verbose)

~/Documents/Projects/MachineLearning/libraries/lib/python3.5/site-packages/mne/utils.py in verbose(function, *args, **kwargs)
    726         with use_log_level(verbose_level):
    727             return function(*args, **kwargs)
--> 728     return function(*args, **kwargs)
    729 
    730 

~/Documents/Projects/MachineLearning/libraries/lib/python3.5/site-packages/mne/utils.py in _fetch_file(url, file_name, print_destination, resume, hash_, timeout, verbose)
   1963                 raise RuntimeError('Hash mismatch for downloaded file %s, '
   1964                                    'expected %s but got %s'
-> 1965                                    % (temp_file_name, hash_, md5))
   1966         shutil.move(temp_file_name, file_name)
   1967         if print_destination is True:

RuntimeError: Hash mismatch for downloaded file /home/krisha/mne_data/MNE-spm-face.tar.gz.part, expected 9f43f67150e3b694b523a21eb929ea75 but got 3eca4e4b6e5341dc2236903136b9d89a

`

larsoner, 2018-03-17:
Are you on latest `master`? It's possible the dataset was updated and you need the latest code to have the correct hash.

krishamehta, 2018-03-17:
I downloaded the dataset directly from the website

larsoner, 2018-03-17:
It looks like `9f43f67150e3b694b523a21eb929ea75` is the correct hash (the one from `master`) so my suggestion won't help.

Can you try downloading using:
```
python -c "import mne; mne.datasets.spm_face.data_path(force_update=True, verbose=True)"
```
I just did this and it succeeded.

larsoner, 2018-03-17:
(you might need to remove the existing `.part` file)

krishamehta, 2018-03-17:
I will try this.  Thanks a lot.

krishamehta, 2018-03-17:
I tried doing it. Got the same error.

`rifying hash 9f43f67150e3b694b523a21eb929ea75.
Error while fetching file https://files.osf.io/v1/resources/rxvq7/providers/osfstorage/59c0e343b83f6902538c4553?action=download&version=2&direct. Dataset fetching aborted.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 2, in data_path
  File "/home/krisha/Documents/Projects/MachineLearning/libraries/lib/python3.5/site-packages/mne/utils.py", line 727, in verbose
    return function(*args, **kwargs)
  File "/home/krisha/Documents/Projects/MachineLearning/libraries/lib/python3.5/site-packages/mne/datasets/spm_face/spm_data.py", line 22, in data_path
    download=download)
  File "/home/krisha/Documents/Projects/MachineLearning/libraries/lib/python3.5/site-packages/mne/datasets/utils.py", line 376, in _data_path
    remove_archive, full = _download(path, u, an, h)
  File "/home/krisha/Documents/Projects/MachineLearning/libraries/lib/python3.5/site-packages/mne/datasets/utils.py", line 435, in _download
    hash_=hash_)
  File "<string>", line 2, in _fetch_file
  File "/home/krisha/Documents/Projects/MachineLearning/libraries/lib/python3.5/site-packages/mne/utils.py", line 728, in verbose
    return function(*args, **kwargs)
  File "/home/krisha/Documents/Projects/MachineLearning/libraries/lib/python3.5/site-packages/mne/utils.py", line 1965, in _fetch_file
    % (temp_file_name, hash_, md5))
RuntimeError: Hash mismatch for downloaded file /home/krisha/mne_data/MNE-spm-face.tar.gz.part, expected 9f43f67150e3b694b523a21eb929ea75 but got ab18152533bbf9b371a98f25dbe8bb04
`

larsoner, 2018-03-17:
That hash is different from the last one you got, which is strange. I don't know what could be causing this. The hash is even listed as `9f43f67150e3b694b523a21eb929ea75` on the osf.io website, consistent with what we have in the code and what I got minutes ago.

You can try downloading it directly from `osf.io` and checking the MD5sum. But so far this indicates to me a problem with your machine, internet connection, or something else local.

krishamehta, 2018-03-17:
I have tried messing around with my server. However, that has not helped either. I will try a few other things.
Thanks!

closed 2018-03-17 by larsoner

AbinS97, 2021-03-07:
I'm having the same error , did you figure it out? if so please help.

larsoner, 2021-03-08:
Did you try redownloading with something like:
```
python -c "import mne; mne.datasets.spm_face.data_path(force_update=True, verbose=True)"
```