mne-tools/mne-python #5028: Using the MNE.decoding.CSP on BCI Competition 3 dataset iii(a)

vivek9746, 2018-03-19:
Hi!! I've been trying to work on the dataset from BCI competition 3 - iii(a).

To achieve the results in a research paper as part of a research program. 

According to the paper**(On Page no 2. (2.2 Pre-processing and feature selection))**, an extended version of CSP, **that is OVR CSP(One Versus Rest Common Spatial Pattern)** is applied to the signals from 60 channels. I don't know exactly how can that be implemented using MNE.

 Can somebody help?

The link to dataset(Just one subject), description and the research paper is: https://drive.google.com/open?id=15PFMvttxvy3SfKXOsELI1SjmjY9UHFXx

kingjr, 2018-03-19:
Hi,

you can simply do:

X = epochs.get_data()
y = epochs.events[:, 2]
csp = mne.decoding.CSP()
cv = StratifiedKFold()

for train, test in cv.split(X, y):
    for category in set(y):
        csp.fit(X[train], y[train]==category)
        y_hat = csp.predict(X[test])

All the best,
Jean-Rémi

vivek9746, 2018-03-19:
Thank you for replying, @kingjr.  :+1: 
Although there was no such thing as **csp.predict()**, I've used **csp.transform()**

On running the above code, I'm getting a an array of some values.

I'm not sure what these are.

Could you please explain the code and the part where **One Versus Rest CSP** is applied?

kingjr, 2018-03-19:
one versus rest consists in finding a hyperplane that separate one category
versus all the other. This is equivalent to make a series of biclass CSPs

the y[train] == category will give you a target vector of indicating
whether each trial belongs to a category or to the rest.

The rest of the procedure is similar to the CSP tutorial/example we wrote:
https://www.martinos.org/mne/stable/auto_examples/decoding/plot_decoding_csp_space.html

Kindest regards,

JR

vivek9746, 2018-03-21:
Thank you again @kingjr :) I have done as you suggested and I have landed with a matrix (354 X 4). I'm kinda stuck now. I'm following a procedure here(Follow the **jpeg.** here: https://drive.google.com/open?id=15PFMvttxvy3SfKXOsELI1SjmjY9UHFXx ) and I'm not sure, how to move ahead from here. If you could refer my code(**3a_k3bData.ipynb**) and suggest necessary addition/edits. I am new to this area and I'd definitely like to move ahead.

Thank you

kingjr, 2018-03-21:
Dear Vivek,

Unfortunately I don't have the resources to guide you through your project.
The aim of the present mailing list is to provide solutions and advice to
concise problems and bugs. I'm sorry not to be more helpful,

Kindest regards,

Jean-Rémi

vivek9746, 2018-03-22:
Thanks for your assistance :+1: :)

closed 2018-03-22 by vivek9746