mne-tools/mne-python #5030: In TF mixed norm solvers, if log_objective=False, the solver runs up to max_it

mathurinm, 2018-03-21:
As @joewalter pointed to me, currently if `log_objective=False`, the gap is not computed:
https://github.com/mne-tools/mne-python/blob/master/mne/inverse_sparse/mxne_optim.py#L946
so `converged` is never set to True and the solver has to run until `maxit` iterations are done.

I believe it would be more natural and flexible to pass `log_objective=n` (integer), and then replace:
```
if log_objective:
    if (ii + 1) % 10 == 0:
```
by
`if (ii + 1) % log_objective == 0:`

This way we offer finer control (e.g.` log_objective=1` fr debugging purposes) and can still pass `log_objective=np.inf` if we want to mimic the current behavior for log_objective=False.

@agramfort if you agree I can do a PR

agramfort, 2018-03-21:
Ok

Although for me dual gap is for early stopping and log objective for debugging and making convergence plots.

joewalter, 2018-03-21:
According to the discussion with @mathurinm, I am in favor of deprecating `log_objective=True/False` and use integers instead. This gives more freedom to the user to select how often the dgap should be evaluated.

joewalter, 2018-03-21:
I found some other small issues (or better options for improvement) in the current master, so I propose to do a small clean up PR with @mathurinm after the PR on dgap has been merged.