mne-tools/mne-python #5033: Wrong axis scaling in .plot_image

sappelhoff, 2018-03-22:
When I call `.plot_image()` on epochs data, the x axis (time) is correctly shown in seconds.

However, when I call `.plot_image()` on evoked data (=`epochs.average()`), the x axis (again, time) is incorrectly shown in milliseconds (although the values indicate seconds).

Having used some of the plotting functions, I have also realized that the labeling for the time axis is not *always* either in milliseconds or in seconds but rather this changes from function to function and apparently depending on the object that I call the function on (see above).

Apart from fixing the issue with `evoked.plot_image()`, maybe it would be better to use consistent labeling for the time axis? Or are there reasons that would speak against this consistency?

Pictures to show what I mean (the data is always the same): 
`evoked.plot_image()` --> time axis label should be "seconds"  ... or `x_axis_values *= 1000`
[image]

`epochs.plot_image()` --> again, this is a correct example ... working with an `epochs` instead of an `evoked` object
[image]

`evoked.plot_joint()` --> another correct example, this time with an `evoked` object again. However why is it in milliseconds now?
[image]

larsoner, 2018-03-22:
> However, when I call .plot_image() on evoked data (=epochs.average()), the x axis (again, time) is incorrectly shown in milliseconds (although the values indicate seconds).

This sounds like a bug we should fix, do you have time to do it?

> Having used some of the plotting functions, I have also realized that the labeling for the time axis is not always either in milliseconds or in seconds but rather this changes from function to function and apparently depending on the object that I call the function on (see above).

Can you comment over on the closed #3164? If enough people gripe it might get @agramfort to budge :)

larsoner, 2018-03-22:
... if you do make a PR, feel free to also take care of this aspect of the plot(s) you change:

https://github.com/mne-tools/mne-python/issues/4534

larsoner, 2018-03-22:
And it looks like there is a bug in that image plot, too. Zero should be white (middle of colorbar).

jona-sassenhagen, 2018-03-23:
I'm fixing the colorer scaling in #5010 already.

sappelhoff, 2018-03-23:
@jona-sassenhagen, that's great. Would be cool if you could incorporate the fix for the axis. 

Making the axis labelling consistent could then be another step as it's not a bug per so :-)

PR #5010 (2018-03-29): MRG evoked.plot_image mask
`evoked.plot_image` gains a "mask" kwarg that takes a boolean. The `False` parts are set transparent. Useful for, for example, quick visualisation of significant tests.

Also fixes the default vlims on `plot_image`, which previously tended to be asymmetric, which is not so good with RdBu_r.

Addresses a part of #2693

closes #5033
closed 2018-03-29 by larsoner