mne-tools/mne-python #5038: Loading data without Raw/Epochs/Evoked

howarth, 2018-03-23:
The gist is that I'd like to know if there exists a way to pull data from a single time point or list of nonadjacent time points given only the Raw object (or just the path) and either the time values or indicies. I.e. I'd like a function load_timepoints(raw, [1.005, 5.024]) that would return two vectors that correspond to the data at time 1.005 and 5.0024.

Our lab uses data averaged over time often. E.g. we'll take the average of a 50ms window. In order to save IO time I'd like to precompute a running sum for each data file. By running sum I mean that the vector for time T is the sum of all values up to T. Then to compute an averaged window you just have to pull in two vectors, subtract first from the second, and divide by the number of time points between them.

I'd like to be able to do this with FIFF files. I know that if I compute the running sum and then save it to a fif file with the same info object as the original FIFF.

I know I can get a single time point through Raw or Epoch's constructor as in

Raw('path/to/fiff').crop(tmin=1.005, tmax=1.005).get_data()

But if I was to do this to compute a sliding window with 10ms overlap I would be making way too many IO calls and also loading thousands of info objects and other metadata each time I get a single vector.

larsoner, 2018-03-23:
> But if I was to do this to compute a sliding window with 10ms overlap I would be making way too many IO calls and also loading thousands of info objects and other metadata each time I get a single vector.

Yep, this will happen. Each `Raw` call (and really `read_raw_fif` is recommended nowadays) will read the `info` from the header.

You can optimize this process, though. Something like this could work instead:
```
raw = mne.io.read_raw_fif(...).load_data()
idx = raw.time_as_index(1.005)  # what index do we need
data = raw[:, idx:idx+1][0]  # makes a copy (in memory) of just the data from this time point
```
If the entire object will not fit into memory but you still need speed, you can use `read_raw_fif(...).crop(...)` to split the data into chunks. (But keep in mind `crop` is inclusive, so you might need to bump the end point by 1 sample's worth of time if you want to do this in consecutive chunks.)

larsoner, 2018-03-23:
>  if there exists a way to pull data from a single time point or list of nonadjacent time points given only the Raw object (or just the path) and either the time values or indicies.

... and to be clear, the answer to this is that you should use `raw[...]` to get these values. It returns a tuple of `data, times`, so doing `raw[:, :][0]` will be all the data points, `raw[:10, 20:30][0]` will be the data from the first 10 channels for time indices 20 (inclusive) - 30 (exclusive), etc.

howarth, 2018-03-23:
I think I found what I was looking for: Raw._read_segment : https://github.com/mne-tools/mne-python/blob/master/mne/io/base.py#L425

That way I only have to have a single Raw object and not create a Raw object each time I call crop(). 

So the real thing I'd want is a _read_segment that reads arbitrary indices instead of segments, all with a single IO, but I'm not even sure if hard disks can do that.

larsoner, 2018-03-23:
Do not use private functions, they can change anytime without warning. `_read_segment` is basically what is called by `raw[...]`, so you should use that public interface.

> I'd want is a _read_segment that reads arbitrary indices instead of segments, all with a single IO, but I'm not even sure if hard disks can do that.

Using the `.load_data()` that I mentioned above loads all time points into memory so no HD I/O will be used anymore.

howarth, 2018-03-23:
Oh... I'm looking at the source and it seems like raw[:, 10] will only read a single timepoint and NOT load the rest of the data, correct?

larsoner, 2018-03-23:
> Oh... I'm looking at the source and it seems like raw[:, 10] will only read a single timepoint and NOT load the rest of the data, correct?

Correct. Depending on how many points you want to load, their spacing, data storage (HD vs SSD) etc., this may or may not be faster than using `.load_data()` followed by the same set of queries.

howarth, 2018-03-23:
Great, thanks!

closed 2018-03-23 by larsoner