mne-tools/mne-python #5046: raw.plot_psd cannot set picks if average is False

moppymopperson, 2018-03-27:
The following throws an exception if `average` is False, but not when it is True. 
The [documentation](https://martinos.org/mne/stable/generated/mne.io.Raw.html#mne.io.Raw.plot_psd) isn't super clear about what the expected behavior in this case is, so I'm not sure if I'm doing something unintended or not.

If `picks` can be specified when `average` is False, then this is definitely a bug. If not, then maybe updating the documentation for clarity or adding a warning message would be appropriate. I'm happy to help with either.

```python
import mne
from mne.datasets import sample

data_path = sample.data_path()
fname = data_path + "/MEG/sample/sample_audvis_raw.fif"
raw = mne.io.read_raw_fif(fname)
picks = mne.pick_types(raw.info, meg=True, eeg=False, stim=False)

# Changing average to True makes it work
raw.plot_psd(picks=picks, dB=False, show=True, average=False)
```

Results in stack trace
```
/miniconda3/envs/mne/lib/python2.7/site-packages/mne/viz/evoked.py:433: VisibleDeprecationWarning: boolean index did not match indexed array along dimension 0; dimension is 1 but corresponding boolean dimension is 2
  for ax in np.array(axes)[selectables]:
Traceback (most recent call last):
  File "psd_bug.py", line 10, in <module>
    ax = raw.plot_psd(picks=picks, dB=False, show=True, average=False)
  File "<string>", line 2, in plot_psd
  File "/home/erik/miniconda3/envs/mne/lib/python2.7/site-packages/mne/utils.py", line 728, in verbose
    return function(*args, **kwargs)
  File "/home/erik/miniconda3/envs/mne/lib/python2.7/site-packages/mne/io/base.py", line 1779, in plot_psd
    xscale=xscale, reject_by_annotation=reject_by_annotation)
  File "<string>", line 2, in plot_raw_psd
  File "/home/erik/miniconda3/envs/mne/lib/python2.7/site-packages/mne/utils.py", line 728, in verbose
    return function(*args, **kwargs)
  File "/home/erik/miniconda3/envs/mne/lib/python2.7/site-packages/mne/viz/raw.py", line 812, in plot_raw_psd
    line_alpha=line_alpha)
  File "/home/erik/miniconda3/envs/mne/lib/python2.7/site-packages/mne/viz/evoked.py", line 433, in _plot_lines
    for ax in np.array(axes)[selectables]:
IndexError: index 1 is out of bounds for axis 1 with size 1

```

agramfort, 2018-03-27:
sounds like an easy pb to solve for a new contributor?

moppymopperson, 2018-04-05:
I can't promise I'll get around to this soon, but if it hasn't been fixed by the next time I've got some spare time I'd definitely like to give it a crack.

PR #5097 (2018-04-05): MRG: Fix picks for raw.plot_psd
Closes #5046.

Using the snippet from #5046:
```
import mne
from mne.datasets import sample

data_path = sample.data_path()
fname = data_path + "/MEG/sample/sample_audvis_raw.fif"
raw = mne.io.read_raw_fif(fname)
picks = mne.pick_types(raw.info, meg=True, eeg=False, stim=False)
raw.plot_psd(picks=picks, average=False)
```
We now get this instead of an error:

[image]
closed 2018-04-05 by agramfort