mne-tools/mne-python #5050: BUG: spoc example broken

kingjr, 2018-03-28:
@agramfort tells me the SPoC example in master is broken, haven't had time to look at it yet.

https://martinos.org/mne/stable/auto_examples/decoding/plot_decoding_spoc_CMC.html

cc @alexandrebarachant

jona-sassenhagen, 2018-03-28:
https://mne-tools.github.io/dev/auto_examples/decoding/plot_decoding_spoc_CMC.html

jhouck, 2018-12-07:
Was this fixed already?  The example works fine for me, although only after updating from mne 0.16.2 to 0.18.dev0 and sklearn 0.19.1 to 0.20.1 .  Prior to the update it didn't like the 'rank' keyword in `make_pipeline` but ran OK without it.  I kept the old version's anaconda environment so can replicate if someone really wants to read the original error message.

dengemann, 2018-12-07:
@jhouck yes we fixed it with @DavidSabbagh. Take a look at current master and let us know if there is some remaining issue.

closed 2018-12-07 by dengemann

jhouck, 2018-12-07:
Thanks @dengemann.  No remaining issues,  I'd borrowed pretty heavily from the example and was worried when I saw the open issue. :)