mne-tools/mne-python #5053: ica.apply zeroing out does not work

EliWim, 2018-03-29:
Hello,
I have a problem I am using ica.apply and try to zero out some of my compnents but it does not work (the result is the same as the original data).  If I use the tutorial (https://www.martinos.org/mne/stable/auto_tutorials/plot_artifacts_correction_ica.html) and I check the data with: 
print(raw._data)
print(raw_copy._data) 
they are different but when I load my own raw .fif file and run the same code I find that 
raw._data == raw_copy._data
Any ideas what may go wrong?

agramfort, 2018-03-29:
we would need to see some code to see what you are doing.

EliWim, 2018-03-29:
data_path = '/nas/b.wutzl/MEG_fMRI/MEG/MEG_tsss_firstround/'
subjects_dir = data_path
raw_fname = data_path + '20180131_1112_LW1_raw_tsss.fif'
raw = io.read_raw_fif(raw_fname , preload=True) 
raw.info['bads']=['MEG0513', 'MEG0613','MEG1041','MEG1111','MEG2013','MEG2233','MEG2243','MEG2643','MEG2926','MEG3125','MEG3244'] 
picks = mne.pick_types(raw.info, meg=True, eeg=False, eog=False, ecg=False, stim=False, exclude='bads')
ica.exclude.extend([1,2,3,4])
ica.apply(raw)
print(raw._data)

raw_pure = io.read_raw_fif(raw_fname , preload=True) 
raw_pure.info['bads']=['MEG0513', 'MEG0613','MEG1041','MEG1111','MEG2013','MEG2233','MEG2243','MEG2643','MEG2926','MEG3125','MEG3244'] 
print(raw_pure._data)

print(raw._data==raw_pure._data)
ica=mne.preprocessing.ICA(n_components=0.9, random_state=0).fit(raw,decim=1,picks=picks)

EliWim, 2018-03-29:
I also followed the tutorial mentioned above and just loaded my .fif file with the same result.

EliWim, 2018-03-30:
I tried now:

ica.exclude.extend([1,2,3,4])
raw_ica=ica.apply(raw)
print(raw_ica._data)

raw_pure = io.read_raw_fif(raw_fname , preload=True)
raw_pure.info['bads']=['MEG0513', 'MEG0613','MEG1041','MEG1111','MEG2013','MEG2233','MEG2243','MEG2643','MEG2926','MEG3125','MEG3244'] 
print(raw_pure._data)

print(raw_ica._data==raw_pure._data)

But I get again that they are that they are equal.

agramfort, 2018-03-30:
please share a full script so we can replicate using idally the sample data.

thank you

EliWim, 2018-04-02:
If I use the sample data as in: 

import mne
from mne.datasets import sample
import os
import glob
from mne import io

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = io.read_raw_fif(raw_fname , preload=True)
picks = mne.pick_types(raw.info, meg=True, eeg=False, eog=False, ecg=False, stim=False, exclude='bads')
ica=mne.preprocessing.ICA(n_components=0.9, random_state=0).fit(raw,decim=1,picks=picks)
ica.exclude.extend([1,2,3,4])
raw_copy = raw.copy()
raw_ica=ica.apply(raw_copy)
print(raw_ica._data)
print(raw._data)
print(raw_ica._data==raw._data)

I get as a result that raw_ica._data is not equal to raw._data but if I just change data_path and raw_fname to my files I get that they are equal.

agramfort, 2018-04-02:
is your data having an ok info?

I am afraid it's hard to tell without looking at  your data

EliWim, 2018-04-02:
what is an ok info?
I cannot load up my .fif file...

agramfort, 2018-04-02:
What do you mean you cannot load the file ?

larsoner, 2018-04-02:
I think they mean they can't upload / share the file

EliWim, 2018-04-03:
Yes exactly i meant I cannot upload it here (so that you could see it...)

agramfort, 2018-04-03:
if you hack your raw with random numbers by modifying the raw._data
attribute do you still have the pb?

EliWim, 2018-04-04:
Yes I do.

agramfort, 2018-04-04:
So I guess now you can share a file so we can debug ?

EliWim, 2018-04-04:
I would like to uploud my file but as mentioned before fif file types are not supported (that is what the error message says).  Is there any other option to share the file  than uploading the file here directly?

agramfort, 2018-04-04:
use dropbox or any other sharing service to share your file.

EliWim, 2018-04-05:
this is my raw fif file:
https://www.dropbox.com/s/56jzrffb2b7sfrn/raw_tsss.fif?dl=0

agramfort, 2018-04-05:
there is something to understand here:

import numpy as np

import mne
from mne.datasets import sample
from mne import io

raw_fname = '/Users/alex/Downloads/raw_tsss.fif'
raw = io.read_raw_fif(raw_fname, preload=True)
raw.pick_types(meg=True)
picks = mne.pick_types(raw.info, meg=True, eeg=False, eog=False, ecg=False,
stim=False, exclude='bads')
ica = mne.preprocessing.ICA(n_components=0.9, random_state=0).fit(raw,
decim=1, picks=picks)
ica.exclude = list(np.arange(30))
raw_ica = ica.apply(raw.copy())
np.allclose(raw_ica._data, raw._data)

even if I remove all ICA components I still get np.allclose(raw_ica._data,
raw._data) that is True....

or maybe I am just tired...

EliWim, 2018-04-06:
I get the same result as you. But I do not understand what is going wrong.

bloyl, 2018-05-08:
Using more reasonable tolerances
`np.allclose(raw_ica._data, raw._data, atol=0, rtol=1e-2)`
I get that the data arrays are different. 

Am I missing something?

EliWim, 2018-05-11:
You are right! Using your option it says that the arrays are not the same - even if they look alike. 
Maybe a stupid question (I am quite new to the field): shouldn't the difference be quite big when zeroing out all components?

jona-sassenhagen, 2018-09-23:
Can this issue be closed?

agramfort, 2018-09-23:
yes it's because values need to be compared in relative difference at numbers are quite small

closed 2018-09-23 by agramfort