mne-tools/mne-python #5054: mne.preprocessing.ICA - components parameters

sappelhoff, 2018-03-29:
I have a question concerning the "component parameters" in ICA, which are `n_components`, `max_pca_components`, and `n_pca_components`. What exactly do they do?

I understand that an ICA of say 64-channel EEG data will usually yield a W of size 64. From my current knowledge the ways how to end up with W of a lower size would be:
1) to perform a dimension reduction with PCA prior to ICA application
2) to "subsample" the sensor space, that is, drop some channels and apply only this subset to ICA

Now the three mentioned parameters to the ICA object are confusing me a bit.

- `max_pca_components` --> this is probably corresponding to my point 1), a dimension reduction and it will select the n components of the PCA that explain the most of the variance (or all components).

If that is true, what is `n_components` doing?

Lastly, what is `n_pca_components` doing? It says that it is:
> The number of PCA components used after ICA recomposition.

... but why would any PCA components be of relevance after I already back-projected my ICA (I assume that is what is meant by recomposition)? Does this parameter perhaps have something to do with the problem that is mentioned in the recent [paper that warns about performing dimension reduction prior to ICA](https://sccn.ucsd.edu/~arno/mypapers/Artoni2018.pdf)?

PS: Asking this question I realized that I don't know how the 64-channel EEG can be reconstructed from an ICA with a W of size e.g. 32 (after pca dimension reduction). Maybe somebody has a quick answer to that as well?

agramfort, 2018-03-29:
for @cbrnr :)

cbrnr, 2018-03-29:
@sappelhoff I completely agree that these arguments are confusing - for an ongoing discussion please see #4856. Here's the tl;dr:

- `n_components` and `max_pca_components` both perform dimensionality reduction via PCA prior to ICA. The *only* difference between these two arguments is the shape of the returned PCA components matrix (with `n_components` you get all components, whereas with `max_pca_components` you get only the selected components, i.e. rows, in this matrix). For the user, I'd recommend to use `n_components`.
- You are right that `n_pca_components` is used in back-projecting. I still haven't figured out why exactly this argument is necessary.

We haven't finished our discussion in #4856 (feel free to participate), but at the very least, we will clarify these arguments in the docs and contrast them to equivalent EEGLAB arguments.

sappelhoff, 2018-03-29:
thanks for the reference, @cbrnr!

Now my current understanding is:

1. calculate PCA on data `X`
2. retain `max_pca_components` and project `X` to `X_reduced`
3. at this point, probably `n_components` is doing something ... but what?
4. Using `X_reduced`, we compute an ICA and have `X_reduced_ica`
5. From `X_reduced_ica`, we retain/reject components
6. We back-project `X_reduced_ica` using the ICA weights and have `X_reduced'`
    - ( the `'` indicating that some components were rejected in the ICA ... but the dimensions are the same)
7. `X_reduced'` can now be back-projected to `X'` using the PCA weights of dimension `max_pca_components`
    - (It would be `X'` instead of the original `X` because of two reasons: (a) some ICA components were missing (rejected) to get the same `X_reduced` and (b) `X_reduced` does not contain all pca components so we cannot recover all the data) 
8. Here, at step 7., `n_pca_components` comes into play by using more than just `max_pca_components` to construct `X'` from `X_reduced'`

Is my intuition about the procedure correct? Given that yes, ... are there any references that suggest that step 8. (concerning `max_pca_components`) is sensible? And I still don't know what `n_components is doing` ... but for now, I will not do any dimension reduction anyway and leave the arguments at None.

However, given the discussion in #4856 and this [paper](https://sccn.ucsd.edu/%7Earno/mypapers/Artoni2018.pdf) that I also linked above, maybe we should include a warning that performing dimensionality reduction (even only by 1% variance) before ICA can lead to unreliable results? That information might fit well next to the [warning note about filters](https://github.com/mne-tools/mne-python/blob/d0f8da7d0f53a7c596b78f22d09197637b9230aa/mne/preprocessing/ica.py#L142-L154) ... ?

closed 2018-03-29 by sappelhoff